betaflight/betaflight-tx-lua-scripts #4: Unify smartaudio not working on lua script

S7ud, 2017-01-28:
I have a MotoLab Cyclone bf3.1. the smartaudio wire from my Unify HV race goes to UART1 TX, sbus is on UART2 and smartport telemetry is on UART3. in ports tab UART one is set to smartaudio. Taranis is running opentx 2.2 RC9. I get the PID  and rate adjustments in the script but no VTX control.

raphaelcoeffic, 2017-01-28:
If you are using release build 3.1.0, it is missing the required support for VTX related MSP requests.

S7ud, 2017-01-28:
Ah OK, that explains it, I thought it made it in. Cheers

rottweilertje, 2017-01-28:
Any specific reason it didn't make it?
I was really looking forward to that. Saves a lot of hassles with annoying hard to reach buttons.

raphaelcoeffic, 2017-01-28:
> 
> Any specific reason it didn't make it?
Too late, RC-Phase was already started.
But it may make it into 3.1.1. If you guys make some noise, that may help 😇

S7ud, 2017-01-28:
@raphaelcoeffic if I open an issue on the betaflight branch will that help? Man, I was looking forward to this feature. 

Is there anyway to test it in the meantime?

raphaelcoeffic, 2017-01-29:
@S7ud you can just use the build from Boris's build server which has been made right after it has been merged (987). Please note that the version will say 3.2 ;-) This is because the version gets incremented right after the official release. Here is a link to that build: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/987/artifact/obj/betaflight_3.2.0-987-bce0415.zip

S7ud, 2017-01-29:
@raphaelcoeffic Thanks, it works. 👍

rottweilertje, 2017-01-29:
Works like a charm on my TBS Unify Pro HV, since it is now marked for 3.2 is there still a possibility for the VTX related MSP requests to make the 3.1.1 patch? And if so, where do we need to make noise to make that happen?

raphaelcoeffic, 2017-01-29:
@rottweilertje wherever you can reach Boris, but you don't have it from me ;-) I already asked, and don't want to bug him twice.

si618, 2017-01-31:
Thanks for heads up @raphaelcoeffic!  I will try a nightly build tonight.  

btw, not sure if this warrants a new issue or not, but are the BFSetup and SpMsp scripts supposed to be able to run side by side on different telemetry pages?

I tried last night, and once I added SpMsp to page 3 telemetry, my BFSetup script on page 2 telemetry stopped showing any pids or rates. They came back once I removed the SpMsp script and power cycled the radio.

raphaelcoeffic, 2017-01-31:
SpMsp is only for testing purposes. I made it to test the link quality over SPORT. You should not install it.

raphaelcoeffic, 2017-01-31:
@S7ud @rottweilertje @si618 You guys might be happy to learn that we will get full support for VTX setup on official BF 3.1.1.

closed 2017-01-31 by raphaelcoeffic

rottweilertje, 2017-01-31:
Nice! Thanks for the work!

S7ud, 2017-02-02:
@raphaelcoeffic Love your work mate.