betaflight/betaflight-tx-lua-scripts #5: Unable to save VTX changes for IRC Tramp

si618, 2017-02-01:
Tramp telemetry is showing on the Taranis, however saving changes is not working for me. 

After "Saving..." is closed the original values are restored, and any modifications are lost. PID and rates are showing and saving correctly.

Hardware:
X-Racer FC v3.1
XSR RX (SmartPort telemetry using [SoftSerial with resistor](https://github.com/betaflight/betaflight/issues/1065#issuecomment-268994219))
Tramp VTX (IRC Tramp telemetry using TX on UART 2)

Software:
ßF 3.2.0 N992
Open TX 2.2 N362
[BFSetup.lua](https://github.com/betaflight/betaflight-tx-lua-scripts/commit/a7741f8ccbd70be76b751d4d62b3137d08ecc082) (renamed to BF.lua)

raphaelcoeffic, 2017-02-01:
Have you tried reloading the page after some time to see what happens? Also, is that happening only when changing multiple items, or is that happening as well when only either frequency or power is changed at the time?

si618, 2017-02-01:
Thanks, I'll try a reload after waiting longer, maybe I'm just being impatient. Will test further tonight.

I did try changing single item, as well as changing multiple items, and it made no difference.

si618, 2017-02-01:
No luck waiting longer after saving before reloading. Not sure if just co-incidence, but one thing I noticed is that channel would sometimes "stick" after saving, so a reload would show change, whereas band, power and pit didn't. 

However, no changes are surviving a power cycle. I am making changes in unarmed state. Also flashed to ßF 3.1.1, but it made no difference.

raphaelcoeffic, 2017-02-01:
@si618 do you have any possibility to test the same with the OSD connected as well? I'm afraid that Mr Tramp can sometimes be quite difficult, when it comes to serial control. So I guess the issue is probably in the BF code controlling the Tramp (which I'm in charge as well, as @jflyper does not own a Tramp).

Just in case, do you own a logic analyser? That would help greatly. In case, just hook it the Tramp while trying out the different scenarios.

Now to the issue itself, here are a couple of questions so that I can better figure out:
- does is the issue always show up, or are there some case where it works, and some not? (please test each value separately, with a power cycle in between, so that we can better isolate each issue)
- when you mean that no changes survive power cycle, do you mean no VTX changes, or do you mean PID/Rate changes as well?

raphaelcoeffic, 2017-02-01:
@si618 could you please tell me what the firmware version printed on your Tramp HV? Please look the corresponding wiki article (updated today with important info).

si618, 2017-02-01:
@raphaelcoeffic Batch 01 v 1.22.

Can also confirm that removing the TNR sensor fixes the save issue for me, yay!

So, I'd say it's 99% fixed; saving of vtx data is working. However, the script sometimes doesn't update the screen with the modified value after saving. Reloading the page updates to the correct value, and it mostly seems to be the "power" setting that is affected by this, although I can't confirm and would need to spend more time testing.

raphaelcoeffic, 2017-02-01:
@si618 great news! 

I think I can come up with a solution for the 1%.

I just need to somehow make sure I get an up-to-date status from the Tramp, and not something in between. The reason why you see the power not being updated is because we have to send separate requests, frequency first, power second, and I have no indication in LUA that the settings update is done.

si618, 2017-02-01:
Sounds great, and I'm happy to test any script changes. 

Thanks again for your excellent work on this, it's such a cool feature :)

raphaelcoeffic, 2017-02-05:
@si618 I've just found a bug preventing power changes. Have a look at betaflight/betaflight#2321.

si618, 2017-02-05:
Cheers!  I'll grab a nightly and test as soon as it's merged into master.

raphaelcoeffic, 2017-02-07:
@si618 The fix has been included in 3.1.4 and should be available for download now.

raphaelcoeffic, 2017-02-07:
strike my last comment, 3.1.5 is the one containing all the fixes. However, you still need to manually reload for now when changing channel & power at the same time.

si618, 2017-02-08:
@raphaelcoeffic looking very good!  Just flashed ßF 3.1.5 and updated scripts on my taranis to 0.1. 

Frequency is working a treat, thank you! As you say, the only thing that appears to not be working reliably is the power value reported back to the taranis after updating both power and channel and the same time. If I do them separately, it seems to work. 

All items appear to be persisting to eeprom reliably, but I will need to spend more time testing to 100% sure. Thanks again, it's such a cool feature! :D

raphaelcoeffic, 2017-02-08:
@si618 very good! I'll try to find a solution to short the waiting and ensure that the final update shows you the final status.

ronerone, 2017-02-08:
Hi, i'm joinging your post.
I'm currently testing the scripts on a taranis in 10.2RC10 and a SP Racing F3 in BF 3.1.5 and XSR.
My T pin is connected directly to TX on UART2.
At the moment, i have no resistor (as i could see above) on my XSR smartport.
Scripts are executing like a charm and on my side, it seems (after few tests) i have to set my channel number and then do a save all plus dearm/rearm to get the value updated on the screen.
I currently have no way to get video feedback so i can't check it actually applies but it sounds good!
I didn't play with the power value as i fear my tramp becomes too hot.
@si618: what did the resistor trick on smartport bring to you? As i'm not sure my configuration is exactly the same, i don't really understand which situation made you apply that trick.

Ronan.

ronerone, 2017-02-08:
as per Raphael, maybe i'm not patient enough and my value would refresh even without the dearm/rearm :)

si618, 2017-02-08:
@ronerone I ran out of uarts, so used soft serial for smart port, and to get that to work properly, I needed to add resistor on rx.  People are getting smart port working on shared uarts, e.g. uart 1 with usb, but you need to use multiwii stick trick to get things to save. See P.Y.'s comment here: http://www.nitbeatfpv.com/frsky-taranis-pid-settings

ronerone, 2017-02-08:
thank you @si618 . In fact, i'm totally newbie and don't know what multiwii stick trick is, gonna have a look.
Currently, i connected my XSR telemetry port to I/O1 TX pin which seems to work.

@raphaelcoeffic , would a LUA script, in addition to yours, to be addressed by a physical switch make sense, if each position could automaticaly set a power mode?
let's say as an example: 25mW on position1, 200mW on position 2 and Pit mode on position 3

That would allow to have personal preferences depending on typical situation quickly set whether your alone in a field or with friends in a race?
Would it be simple to you to export from your set of scripts?

I have no knowledge in the programming language used in LUA at the moment so i can only suggest :)

raphaelcoeffic, 2017-02-08:
@ronerone the power-change switch will be addressed directly in betaflight. I believe it's been developed right now. This is definitely better, as it will work for everybody, and allow you to configure the switch itself directly in betaflight configurator.

ronerone, 2017-02-08:
Excellent piece of news! les grands esprits se rencontrent ;)
After few tests, i confirm the behaviour on my side:
1/ set the PID value
2/ save all
3/ unarm and rearm

If i don't execute step 3, i have no refresh on the screen, but maybe it's actually applied, which i can't check now.

raphaelcoeffic, 2017-02-08:
@ronerone aha! So you have SmartPort shared with MSP, which means that it will only work when armed, but the settings cannot be saved while armed (blocked on the FC) ;-) Be patient, someone is working hard on softserial for the Tramp/TBS unify, so that you should able to move SmartPort away from UART1. Or you just buy yourself a new FC with 3 UARTs and USB using VCP (Virtual com port, not blocking any of the other UARTs).

ronerone, 2017-02-08:
@raphaelcoeffic ahah! i just finished mounting it and internal agency is probably the worst part of the job :)
I don't feel good thinking of restarting everything, i will wait for the developper 💃 
Anyway, knowing how to proceed is 50% of the issue solved afaik, and it's not a big deal, it still easier than pressing the small hidden button, missing the sequence, restart, and so on eheheh!

closed 2017-11-14 by raphaelcoeffic