betaflight/betaflight-tx-lua-scripts #8: Change file name to a shorter one

nitbeat, 2017-02-02:
Hello,

Why not change the filename to max 6 charechters ?
Some people forget doing that and it doesn't work properly.

Thanks :)

si618, 2017-02-03:
Just call it BF.lua

Problem solved!

raphaelcoeffic, 2017-02-07:
@nitbeat @si618 Gents, I'm 90% with a brand new version which has the common parts between the different TXs split, so that the code is not duplicated and better organized. We will have the main scripts called after the TX-type in the repos, but the individual packages for the targets (X7, X9, Horus) will have BF.lua set as the script name.

si618, 2017-02-07:
Sounds great, love your work Raphael!

raphaelcoeffic, 2017-02-07:
implemented in #11

closed 2017-02-07 by raphaelcoeffic

raphaelcoeffic, 2017-02-07:
see https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/v0.1