betaflight/betaflight-tx-lua-scripts #9: Adding Favorite Channels option

nitbeat, 2017-02-02:
**Feature Request** 

In my country we use the same 4 frequencies every time we fly multiple pilots, it's always:
F1, F7, E4, E8
We learned these work together pretty well most of the time.

I'm suggesting to add an option to scroll through channels that were selected to be favorites,
so I can just select them without changing band and freq for specific channel.

It's a luxury I know, but could be nice.
@raphaelcoeffic

raphaelcoeffic, 2017-02-02:
That probably need some thoughts to be put in the UI. And that could require some kind of config file (probably in LUA, or just editing the main script?).

johnryanlee, 2017-05-07:
Even better would be to be able to use a switch to set a frequency.  So a 3 pos switch would set the VTX to 3 possible frequencies.  4 would be ideal when racing as we race with 4 pilots at a time here.

closed 2018-05-13 by stale[bot]