betaflight/betaflight-tx-lua-scripts #20: TBS Unify Pro HV - L Band issue

mhenrichsen, 2017-03-03:
Hi,

I noticed if I start my Unify on the L band I get an error and won't be able to access the vtx window.
This can be fixed by adding the channels manually to the ui.lua, but you're not able to swap channel to any of them.

Can you fix this, so I can change to the L band?
5362, 5399, 5436, 5473, 5510, 5547, 5584, 5621

raphaelcoeffic, 2017-03-06:
@mhenrichsen The issue is not just located in the script, but in BF as well. Once they have been added to BF, we will use them in the script as well.

matbaig, 2017-05-08:
Would love it if this was possible! In my country I am legally allowed to use this Band but unfortunately dont run osd on my quads so no betaflightf3 or any other possible form of osd to change to this band.

AJZOMBIE, 2017-09-19:
72 Channal

---- File BF.lua----
{
      title = "VTX",
      text = {},
      fields = {
         -- Super Rate
         { t = "Band",    x = 25,  y = 14, sp = 50, i=2, min=1, max=9, table = { "A", "B", "E", "F", "R" , "L" , "U" , "O" , "H" } },
         { t = "Channel", x = 25,  y = 24, sp = 50, i=3, min=1, max=8 },
         { t = "Power",   x = 25,  y = 34, sp = 50, i=4, min=1 },
         { t = "Pit",     x = 25,  y = 44, sp = 50, i=5, min=0, max=1, table = { [0]="OFF", "ON" } },
         { t = "Dev",     x = 100, y = 14, sp = 32, i=1, ro=true, table = {[3]="SmartAudio",[4]="Tramp",[255]="None"} },
         { t = "Freq",    x = 100, y = 24, sp = 32, i="f", ro=true },
      },
---- msp.lua
local freqLookup = {
    { 5865, 5845, 5825, 5805, 5785, 5765, 5745, 5725 }, -- Boscam A
    { 5733, 5752, 5771, 5790, 5809, 5828, 5847, 5866 }, -- Boscam B
    { 5705, 5685, 5665, 5645, 5885, 5905, 5925, 5945 }, -- Boscam E
    { 5740, 5760, 5780, 5800, 5820, 5840, 5860, 5880 }, -- FatShark F
    { 5658, 5695, 5732, 5769, 5806, 5843, 5880, 5917 }, -- RaceBand
	{ 5362, 5399, 5436, 5473, 5510, 5547, 5584, 5621 }, -- L Band
	{ 5325, 5348, 5366, 5384, 5402, 5420, 5438, 5456 }, -- U Band	
	{ 5474, 5492, 5510, 5528, 5546, 5564, 5582, 5600 }, -- O Band
	{ 5653, 5693, 5733, 5773, 5813, 5853, 5893, 5933 }, -- H Band
}

closed 2018-05-13 by stale[bot]