betaflight/betaflight-tx-lua-scripts #21: BF 3.1.6 PIDs not shown, just ---

RipperDrone, 2017-03-05:
Had an older script BFSetp.lua running ok, showed & saved values.

Now updated to BF 3.1.6, latest release of lua scripts, installed BF and TELEMETRY directories under SCRIPTS subdir of my Taranis X9D+. Executed script, mapped it onto screen 2 of telemetry pages.

PID page shows up, but no values, only ---

I have latest RC11 OTX 2.2 firmware on my Taranis.

Any thoughts?

FPWibi, 2017-03-05:
I have the same Problem. But it is definitely a problem with OTX RC11. When I change back to RC10 it is working.

RipperDrone, 2017-03-05:
I see, so we need to wait for another patch or stay with RC10... :-(

FPWibi, 2017-03-05:
I don't really know anything about it, but there have been the following changes with the RC11 release with LUA scripts:

LUA : all numbers are now handled as float instead of double as previously (speed and memory footprint improved)
LUA : system enhancement have been made to improve stability of scripts

The code of the LUA script maybe has to be changed.

RipperDrone, 2017-03-05:
I have tried to find a response on OpenTX github, but they replied that the author of the script should check where the error comes from, because they do not seem to be deeply involved in these specific scripts at OpenTX. 

Looks like the BF side might have to analyze it further if we want to get it resolved...

Professorvennie, 2017-03-05:
I am having the same issue.  Would it be possible for me to downgrade to rc10?

FlowBowFPV, 2017-03-05:
If anybody need the RC10 for X7 or X9d+
https://www.dropbox.com/sh/9dc2pu3u0uz66gk/AAClI98k9YrZnlZ88M9x6Ajaa?dl=0

Greez 
FlowBow

RipperDrone, 2017-03-05:
Got these hints from OpenTX GitHub: 

From the script it is not evident which dataId is used when sending the MSP frames. Somebody should check in betaflight source. The valid range now is:

#define DIY_STREAM_FIRST_ID       0x5000
#define DIY_STREAM_LAST_ID        0x50ff
See d594843

Check out details here, maybe it helps: https://github.com/opentx/opentx/issues/4560

RipperDrone, 2017-03-05:
not sure about the details, but I tried to find the snippet where dataID for temeletry puh packet is defined in lua script:

local function mspSendSport(payload)
 +
 +   local dataId = 0
 +   dataId = payload[1] + bit32.lshift(payload[2],8)

Hope it helps a little - @raphaelcoeffic: Will these values be btw 0x5000 and 0x50ff at all times?

mikeller, 2017-03-05:
I think @lshems makes a good point in opentx/opentx#4560 about integer constants getting messed up by conversion to float. The calculations might have to be rechecked to make sure they are not affected by rounding.

raphaelcoeffic, 2017-03-06:
@RipperDrone look at my comment on the same opentx issue: we do not use the dataID at all ;-) Instead, we use a back door on the packet type, thus letting us use the dataID as part of the payload. I've checked the code and this part has not changed. So I tend to believe the issue is related to the number format that has changed (double -> float), thus messing up the binary compatibility for integrers passed.

barthalamu, 2017-03-10:
How can I get RC10 firmware? When I try to download from within Companion RC10, it keeps downloading RC11... I'm using an original (non-plus) version of the Taranis.

teramax, 2017-03-10:
same problem, at least found a zip file with the "bin" here:
http://blog.seidel-philipp.de/frsky-taranis-q-x7-firmware-flashen-eu-lbtnon-eustock/

voipmonkey, 2017-03-10:
Guys 
I had same issue was a model memory issue remove the model and re create it should resolve the issue
Mark

Shadow-MasterTf, 2017-03-13:
RC12 is out guys should fix this issue http://www.open-tx.org/2017/03/13/opentx-2.2.0RC12

Matthias7Song, 2017-03-15:
OTX 2.2.0-RC12 / Script Syntax Error. 
Screen shows as below.

Srcipt syntax error
attempt to call a nil value

closed 2017-04-20 by RipperDrone

woojak, 2019-01-13:
Hello, I had the same problem with my SPRACINGF3EVO.
The old software was beautiful, without problems (3.2.4 Jan 10 2018). When I updated to the latest version of Beta flight, the LUA script worked but did not show the value.
After returning to 3.2.4 LUA started working. I compared dump with both software, but I found only 2 differences and I think that they have no effect. Following the logic, just something is changed in the BF software and that was a problem in my case.
I hope that my observations will be useful:]
greetings