betaflight/betaflight-tx-lua-scripts #22: Great but missing some futures

Dutchy-79, 2017-03-17:
This is really great. 

Only thing is I miss the option to show profile 1 2 and 3.

And when switching to another profile with the sticks the page isn't updated, you need to change page and back to update it. 

I'd like to see it showing the selected profile and update the numbers when changing profile. 

Regards,

beeb, 2017-09-06:
Additionnally, I think it would be nice, if technically feasible, to be able to switch between rate profiles (on top of pid profiles). Cheers

mikeller, 2017-09-06:
@beeb: Just in case you are not aware: It's possible to configure rate profile switching onto a 3 position switch through the 'Adjustments' tab, and then switch in-flight.

beeb, 2017-09-07:
@mikeller absolutely! The only problem is that I have no free channel on my RX for this (all occupied by flight mode, buzzer, reverse dshot direction and arm switch). I could re-bind my RX with 16 channels but I heard it increases the latency slightly over 8 channels.

mikeller, 2017-09-07:
@beeb: There are plans to address this shortcoming in 3.3. Watch this space. ;-)

closed 2022-02-04 by klutvott123