betaflight/betaflight-tx-lua-scripts #28: TBS crossfire

pmijoiu, 2017-05-14:
are there any plans to make this feature over the TBS Crossfire telemetry system?

mikeller, 2017-05-14:
Nobody has come forward yet and offered to implement it. :-)

I am not sure if TBS Crossfire allows data that is unrelated to RC control / telemetry to be sent bidirectionally over the link, so possibly it's not even feasible.

pmijoiu, 2017-05-15:
I assume that the crossfireTelemetryPop() and crossfireTelemetryPush() LUA functions may be used in a similar manner to the way the sPort telemetry functions are currently being used.

I am willing to help adapting the LUA scripts but someone familiar with the corresponding feature in the betaflight firmware must do the actual hard work.

mikeller, 2017-05-15:
@pmijoiu: According to https://rcdiy.gitbooks.io/test/content/general/crossfireTelemetryPop.html this support seems to be quite preliminary (including the packet being called 'SPORT packet'). Looking at the crossfire implementation, it should be possible to inject extra packets into the stream of RC commands and telemetry answers with a distinct frame type, as long as the hardware does not filter these.

I am happy to help with the implementation, but without access to hardware development / testing I am not in a good position to do the implementation myself.

fedorcomander, 2017-05-29:
Hi guys!

Since kiss scripts are based on betaflight ones, i wanted to share

https://www.youtube.com/watch?v=O8uEa9K52B0

You can check my repository how its done.

https://github.com/fedorcomander/kissfc-tx-lua-scripts

Only one think i've noticed is that in latest opentx (use rc18 atm) - receiving packet takes longer then on previous versions. I think sport has same delays. 

Commands 0x78 and 0x79 are reserved for KISSFC, but i think BF can happily use it as well ;)))

CF 1.65 beta required!!!

fedorcomander, 2017-05-29:
1.65 will be added to public beta soon

pmijoiu, 2017-05-31:
Great stuff!

peteoz, 2017-07-21:
the work that's been done on this by Alex to make the scripts working with Crossfire, that's only for KISS FC, correct ?
is anyone developing a version to make it work with Crossfire on Betaflight FCs ?

flyduino, 2017-07-21:
peteoz: lua is rather simple language, why not to try yourself? ;)

fedorcomander, 2017-07-21:
#peteoz LUA is quite simple language, date to try yourself? ;))

peteoz, 2017-07-21:
I understand it's easy for you guys :) , probably not that easy for an average bloke. 

so what you are saying - only writing lua script is all that's required to get the Crossfire to communicate with BF FC ? 
can you give me any pointers where to start and/or what lua units require modifications ?

thanks

TempleClause, 2017-08-08:
@fedorcomander So only LUA Scripts are missing to make this work with CRSF? I could possibly write them. I have no plan about LUA but I'm a software developer, so I might be able to do it. 
Is there also something that has to be done on the BF firmware or only on the LUA Scripts?

fedorcomander, 2017-08-08:
No. Firmware support required as well.

umbertoghio, 2017-08-09:
Hi, I have bought the new crossfire "micro", as it is becoming cheaper (120$ for TX and RX) I think more people will need this awesome feature, I am a Java developer with some C++ background and have already built Betaflight from source to test some code customisation, I am willing to help (testing, debug, coding), but as Peteoz noted I also need some pointers to where to start. Thanks!

peteoz, 2017-10-14:
I got the Lua scripts working with CRSF on my Taranis x9d+ following this guide
https://github.com/codecae/betaflight-tx-lua-revised

there is also a video
https://www.youtube.com/watch?v=BxC_Ac4ljD4

codecae, 2017-11-14:
These scripts now support Crossfire!  For best results, use Betaflight 3.2.2 and Crossfire v2.11 or greater

closed 2017-11-14 by codecae