betaflight/betaflight-tx-lua-scripts #30: Ersky9x - LUA

MikeBland, 2017-06-05:
I've just added some LUA support into ersky9x. This is available on most platforms Taranis/Plus/QX7, 9X with SKY/AR9X/9Xtreme boards, 9XR-PRO. The problem with some of these is limited RAM. Also, they use different navigation buttons.
I have sending and receiving SPort packets working, so it might be possible to implement some of these scripts (1 page at a time?), provided they can be cut down to fit the available memory.

Mike.

mikeller, 2017-06-05:
That is great news, thank you very much @MikeBland!

Do you think you will be aiming to make the ersky9x lua API compatible with the one that is implemented in OpenTx, so that the existing OpenTx scripts can be used, or will it be a different API?

If it is a different API, I would suggest we move the existing (OpenTx) contents in this repo into a `opentx` directory, and add another directory `ersky9x` for the ersky9x scripts.

MikeBland, 2017-06-05:
The API is much the same (as far as I have things implemented), but there are some differences in general, e.g. ersky9x doesn't have the "small font".
I'm posting progress (with API) here: http://openrcforums.com/forum/viewtopic.php?f=7&t=10230#p128831
Mike.

DieHertz, 2017-06-05:
So cool, thank you Mike. Now my 9xtreme doesn't need OpenTX anymore.

DieHertz, 2017-06-06:
I think we should resort to code-generation for OpenTX and ErSky9x, or just maintain two separate versions for each. If the only difference is button codes and display size, some sort of preprocessing to abstract these out would be beneficial.

mikeller, 2017-06-06:
Agreed, considering the API is the same. It could probably also be solved by dynamically loading the parts that are different between OpenTx and ersky9x, and the user uploading the respective 'driver' files for their hardware.

MikeBland, 2017-06-06:
LUA does seem to be very RAM use intensive. The 9XR-PRO has about 22K of RAM available for LUA, but this seems to get used up very quickly.

Mike.

DieHertz, 2017-06-06:
There should be a way to optimize it, it's probably not written the best way possible given the constraints of RAM on MCU.
If that doesn't help, we could integrate TinyScheme header-only interpreter or something similar with small RAM footprint to make it more viable.

raphaelcoeffic, 2017-11-14:
@MikeBland the revised code that has been now merged into this repos should use less memory, as it loads only one page at a time. Would you please try it again and tell us?

MikeBland, 2017-11-14:
Due to the RAM usage of LUA, I've chosen to implement a different script language, loosely based on the "Basic" language. Details are on this thread: http://openrcforums.com/forum/viewtopic.php?f=7&t=10325.
This uses much less RAM than LUA, but does need scripts to be written specifically for it.
I have rewritten the FrSky LUA script for configuring a S6R/S8R receiver and that works fine.
In the thread above, someone is trying to write a betaflight PID editor in this script language. Not having any betaflight hardware, I can't help much.

DieHertz, 2017-11-14:
Hey MikeB, I've been talking about you and your scripting language literally a minute ago

raphaelcoeffic, 2017-11-15:
@MikeBland hey Mike! I guess we can close that one, right?

closed 2018-05-06 by MikeBland