betaflight/betaflight-tx-lua-scripts #31: Not seeing PIDS or Vtx

Kineticheli, 2017-06-09:
Hello, my issue is that PIDs and Vtx are not displayed on my screen. I know my open tx is and older release, so I will update it tonight and see is that helps. Also I have read that VSP shares a uart and only works when the board is armed, I'll try that tonight also. If anybody can think of anything else, I would appreciate  any suggestions.

Hardware: Radiance Dshot board
                  Xsr s. port to tx2
                  s. bus to rx3
                  Tramp Hv tx1
Software: BF 3.1.7
                open Tx RC13

8bitreid, 2017-06-09:
I see this also.

digaus, 2017-06-15:
@Kineticheli 
PID should show up even when disarmed since your Telemetry is connected to TX2. Did you activate Smartport on that UART. Also did you set telemetry inversion to ON and activate telemetry in the configurator tab? Looks like you do not get any telemetry values.

digaus, 2017-06-15:
Might also be a bug of your reciever: https://github.com/betaflight/betaflight/issues/3223

bbc25185, 2017-06-15:
This LUA code work only in OpenTx 2.2.0....
Older version of OpenTx can not work!!!!

Dutchy-79, 2017-06-15:
The Lua script worked on the release candidates also, except for 2.2RC11

Ive used it in RC13, 14, 15 and RC18

biacz, 2017-06-25:
i have to back this up. 

not seeing any pids either.
latest xsr version (6/7/17)
latest opentx version (2.2.0)
betaflight (spracingf3) 3.1.7

telemetry is working fine, just not getting pids.

Dutchy-79, 2017-06-25:
Same here with EU-LBT XSR (6/7/17)
X9D SE (2.2.0)
Betaflight 3.1.7 Matek F405

Telemetry working fine on SoftSerial1 (Inverted signal)

biacz, 2017-06-26:
just downgraded the xsr firmware to the 2015 version. works now. strangely

Dutchy-79, 2017-06-27:
There's a new firmware available for the XSR, I can confirm that it works fine now.

Kineticheli, 2017-07-11:
Finally got around to updating the xsr firmware, all works now. Thanks for all help.

RipperDrone, 2017-07-17:
latest firmware working!

jangrewe, 2017-07-19:
Joshua posted the link to this firmware on FB, saying that it fixes it. (I can't test, i have an X4R-SB)

http://www.frsky-rc.com/download/view.php?sort&down=391&file=Firmware-XSR

piepluquad, 2017-08-18:
may that help someone : using softserial on spracingf3 for smartport telemetry, you need to set both softserial telemetry output baud rate to the same value (or it will loose telemetry in few seconds), also i updated to latest XSRLBT170619 firmware for my XSR

beeb, 2017-09-06:
I can confirm that upgrading to the 170619 firmware solved the issue

closed 2017-11-14 by raphaelcoeffic