betaflight/betaflight-tx-lua-scripts #38: Max Vel (deg/s) in Rates Screen

PilotMan87, 2017-08-31:
Not sure how that number is created, but if that would be available, would be great.

[image]

beeb, 2017-09-06:
For reference, this is how it's computed by betaflight configurator:

    RateCurve.prototype.rcCommandRawToDegreesPerSecond = function (rcData, rate, rcRate, rcExpo, superExpoActive, deadband) {
        var angleRate;
        if (rate !== undefined && rcRate !== undefined && rcExpo !== undefined) {
            if (rcRate > 2) {
                rcRate = rcRate + (rcRate - 2) * 14.54;
            }

            var inputValue = this.rcCommand(rcData, rcRate, deadband);
            var maxRc = 500 * rcRate;
            
            var expoPower;
            var rcRateConstant;
            if (semver.gte(CONFIG.apiVersion, "1.20.0")) {
                expoPower = 3;
                rcRateConstant = 200;
            } else {
                expoPower = 2;
                rcRateConstant = 205.85;
            }

            if (rcExpo > 0) {
                var absRc = Math.abs(inputValue) / maxRc;
                inputValue =  inputValue * Math.pow(absRc, expoPower) * rcExpo + inputValue * (1-rcExpo);
            }

            var rcInput = inputValue / maxRc;

            if (superExpoActive) {
                var rcFactor = 1 / this.constrain(1 - Math.abs(rcInput) * rate, 0.01, 1);
                angleRate = rcRateConstant * rcRate * rcInput; // 200 should be variable checked on version (older versions it's 205,9)
                angleRate = angleRate * rcFactor;
            } else {
                angleRate = (((rate * 100) + 27) * inputValue / 16) / 4.1; // Only applies to old versions ?
            }

            angleRate = this.constrain(angleRate, -1998, 1998); // Rate limit protection
        }

        return angleRate;
    };

mikeller, 2017-09-06:
@beeb: For reference, the implementation in the configurator was originally created by hand-translating the calculations that are done in the firmware into JavaScript. Since the firmware has been updated a number of time since then, and configurator not, it's not really accurate any more. Probably better create a fresh lua translation from the firmware.

beeb, 2017-09-07:
@mikeller good to know! That would indeed be a nice thing to do.