betaflight/betaflight-tx-lua-scripts #39: Low RaceBand change TBS Race HV

LeinesFPV, 2017-09-01:
Hello,
I have changed the script to be able to set LowRaceBand. (TBS Unify HV Race)
Unfortunately, he does not save the settings. He jumps back to the old settings.

Here are my changes:

`Quellcode
fields = {
         -- Super Rate
         { t = "Band",    x = 25,  y = 14, sp = 50, i=2, min=1, max=6, table = { "A", "B", "E", "F", "R", "LRB" } },
         { t = "Channel", x = 25,  y = 24, sp = 50, i=3, min=1, max=8 },
         { t = "Power",   x = 25,  y = 34, sp = 50, i=4, min=1 },
         { t = "Pit",     x = 25,  y = 44, sp = 50, i=5, min=0, max=1, table = { [0]="OFF", "ON" } },
         { t = "Dev",     x = 100, y = 14, sp = 32, i=1, ro=true, table = {[3]="SmartAudio",[4]="Tramp",[255]="None"} },
         { t = "Freq",    x = 100, y = 24, sp = 32, i="f", ro=true },
},`

`Quellcode
local freqLookup = {
    { 5865, 5845, 5825, 5805, 5785, 5765, 5745, 5725 }, -- Boscam A
    { 5733, 5752, 5771, 5790, 5809, 5828, 5847, 5866 }, -- Boscam B
    { 5705, 5685, 5665, 5645, 5885, 5905, 5925, 5945 }, -- Boscam E
    { 5740, 5760, 5780, 5800, 5820, 5840, 5860, 5880 }, -- FatShark
    { 5658, 5695, 5732, 5769, 5806, 5843, 5880, 5917 }, -- RaceBand
    { 5362, 5399, 5436, 5473, 5510, 5547, 5584, 5621 }, -- Low Race Band
}`

Have I forgotten something?

Many Thanks!

raphaelcoeffic, 2017-11-14:
@LeinesFPV I think the issue is in BF, but we can add the low band as soon as BF supports it.

codecae, 2017-11-14:
There have been discussions about adding low band to Betaflight, but it has led to some debate around legality, etc.

raphaelcoeffic, 2017-11-14:
@codecae no kidding?? quite some channels from the normal band are not legal here in europe, but still available in that list...

mikeller, 2017-11-14:
The whole discussion about the legality of frequencies is a hard one - there probably isn't one of the 5.8 GHz frequencies that is _not_ illegal somewhere. But on the other hand, I feel that the discussion has to be had, and an effort has to be made to identify ways to help users be compliant - if only to then conclude that there is no practicable way for Betaflight to do this, and be confident in saying that we did our best to do so.

m3morx, 2017-11-28:
looking forward to see this in the new BF releases. i'm looking for the same feature.

klutvott123, 2022-02-03:
I don't think this is needed now that we have vtx tables.

closed 2022-02-03 by klutvott123