betaflight/betaflight-tx-lua-scripts #40: Feature request: RSSI injection

mikeller, 2017-09-18:
Considering that most FrSky RX make it hard for the user to get RSSI information into the flight controller (extra wire and ADC input required), how about we add an 'RSSI reflector' into the TX lua scripts?
i.e. we add an MSP_SET_RSSI call to the firmware, and make the lua script on the TX periodically send the RSSI value, for display in the OSD. I am aware that this is to some extent the tail wagging the dog, but I think it is a 'better than nothing' alternative for configurations where other means to get the RSSI are not available.

(The TX side of this should be on/off configurable, so that in cases where the RSSI is fed in on the flight controller the injected values do not override the local values.)

nmaggioni, 2017-09-18:
If your TX already gets telemetry, why not just feeding the RSSI value into an empty RC channel/mixer?

https://oscarliang.com/rssi-ppm-channel-taranis/

mikeller, 2017-09-18:
@nmaggioni: That's another option, for sure. But it has the downside of requiring an entire RC channel for RSSI. Especially when running FrSky X with 9ms frame interval (8 channels) that's a high price to pay for RSSI, if it can be done with MSP without requiring a dedicated channel.

Arakon, 2017-10-18:
Also, the RSSI readout using the extra channel is pretty delayed, it doesn't update as fast as it does in a direct RSSI link.

0crap, 2018-08-29:
I think this one can be closed because it's now very easy for the user to get RSSI information into the flight controller.
The SmartPort Telemetry wire is all you need.
See BF [issue 4507](https://github.com/betaflight/betaflight/pull/4507).

On my setup I can confirm it works beautifully.

mikeller, 2018-08-30:
Thanks @0crap, I had totally forgotten about this one.

closed 2018-08-30 by mikeller