betaflight/betaflight-tx-lua-scripts #43: R-XSR not showing PIDS or VTX (Telemetry working ok)

voipmonkey, 2017-09-26:
Need a little help guys. I have a new R-XSR and telemetry works back to radio ok but no Pids or VTX are showing in the screens.
I do have this on softserial if this make a difference not sure.
FC is Matek F405 and all firmwares on everything are at the latest.
Also this works on all my other models just not the new one i am building.
Thanks
mark

Dutchy-79, 2017-09-26:
On the transmitter do you have a sensor called F000

voipmonkey, 2017-09-26:
Yes I have f000

Dutchy-79, 2017-09-26:
That means that the R-XSR is on the right firmware for 2 way communications.

I'm pretty sure the Matek F405 is capable of 2 way on the softserial S5 pad (if you used that like the manual tells you)

miqueljardi, 2017-09-27:
I have the same issue using:

R-XSR Non-Eur
FuriousFPV Radiance F3

Telemetry is working fine, but no PID, rates, vtx... information appears in BF Script Telemetry Screen.

I think it's not FC related or build related issue...

voipmonkey, 2017-09-28:
Right guys transferred to a xsr and all working perfect so must be firmware on new rxsr maybe ??

Dutchy-79, 2017-09-29:
Frsky had the same problem with a previous XSR firmware, just email frsky and let them know.

RipperDrone, 2017-09-30:
On their website, FW 170904 is available. Have you verified with this version?

voipmonkey, 2017-09-30:
Yes this is the version I tried also tried let and fcc both don’t work. This is the only few on there site. Waiting for response from FrSky support

jdpworks, 2017-10-01:
Having the same problem with lua scripts but Kiss FC. Must be r-xsr firmware.

isilcala, 2017-10-01:
Same here. R-XSR on softserial(S5 PAD per Matek instruction). Smartport telemetry works fine but PID / VTX don't work. F000 present. 
@voipmonkey you emailed Frsky?

voipmonkey, 2017-10-01:
@isilcala I have but still waiting on reply maybe we need more emails to frsky

RipperDrone, 2017-10-01:
I've sent a query to FrSky support, too :-)

jdpworks, 2017-10-01:
This from FRSKY:

Have checked with R&D team. The new firmware is in the final testing process. Most likely will  upload next week. Thanks!

RipperDrone, 2017-10-01:
Fabulous - thanks a lot. Your close interaction with the dev group of BF is reamrkable! Thumbs up! :-)

isilcala, 2017-10-03:
@voipmonkey @jdpworks @RipperDrone Thank you all.
China is in National holidays this week so response could be delayed. Looking for next week or so.

aukrausmi, 2017-10-08:
while BF and KISS LUA scripts dont work full stop, interestingly the RF LUA partially works but only on PIDs and not SmartAudio - makes me think that the R-XSR firmware is using a differnt baud rate form the other RXs like XSR & X4R - can't wait for the fix so I can test on all the various FC / VTx / Rx combos I have

voipmonkey, 2017-10-09:
Had a response from frsky and they have sent me some code to test so will get this tested today and feedback hopefully this will be made live soon if it works :)

Fratack, 2017-10-09:
I'm waiting for your test... hope it works

Dutchy-79, 2017-10-09:
The beta file is available for download on the website, not a secret ;)
https://www.frsky-rc.com/r-xsr/

Fratack, 2017-10-09:
I've burned an R-XSR because i've updated the firmware with the servo cable provided, but it has inverted polarity :( anyways Thanks

Dutchy-79, 2017-10-09:
The pinout on a X9D is inverted.

Dutchy-79, 2017-10-09:
Maybe you can replace the voltage regulator. Or power it off 3.3v on the regulator output.

Fratack, 2017-10-09:
Oh Yes.. I've to try it... Can you share some pictures about the Regulator and the right Pin OUT of x9d? Thank you Man.

voipmonkey, 2017-10-09:
love it that firmware was not on then when I looked this morning ... Thanks frsky will still report back

goebish, 2017-10-09:
I confirm it's working again with version 171009

voipmonkey, 2017-10-09:
sweet thanks, goebish I will close this issue down then :) thanks all for the conversation nice to get this cleared up quickly

closed 2017-10-09 by voipmonkey

RipperDrone, 2017-10-09:
so sweet, I love it!

controlflow, 2017-12-15:
please note, that newer R-XSR firmware version — 171103 — is a LBT firmware only and the bf lua script still do not works, FCC version of 171009 works just fine

ScottDunk, 2018-01-16:
I cant get this working and its doing my head in Matek f405 not AIO and r-xsr, got telemtry but no values showing, have tried LBT and FCC versions of 1009, only LBT will bind as that matches my X9D+.

tried LBT version of 103 and thats not showing any values either.

have confirmed in BF that all is set correct

naitse, 2018-01-23:
I'm having issues too, FCC 171009 not showing pids though I have RxBT, F000 and RSSI displayed at telemetry page :/ any idea on how to debug? is there any debug log somewhere?

mikeller, 2018-01-23:
@naitse: There were some changes in the SmartPort code to fix timing issues with r-XSR. Can you try a 3.3 nightly build to see if that fixes it for you?

naitse, 2018-01-24:
@mikeller no luck :/, I tried inverted an un-inverted telemetry only difference I see is that F000 gets weird values, some times 9, 16, 199 XD but no PIDs

michael-blue2, 2018-01-24:
So guys, i did installed yesterday the r-xsr to a build of mine. Kalute f4 taranis x7. previeus i was using an x4r sb, all perfect. As soon as i got the r-xsr i uodated to the 171009 as i am using an international version. i can get telemetry BUT no f000. Actually i never rember i had f000 on other builds.

michael-blue2, 2018-01-24:
i also tried univerted and inverted telemetry with all the settings in bf. for inversion and halfduplex( if i remeber correct)

naitse, 2018-01-24:
did you used the un-inverted pad for telemetry on the r-xsr? that solved the problem for me, also I had to restart the taranis after doing that to see the changes [https://oscarliang.com/ctt/uploads/2017/10/frsky_r-xsr_rx_receiver_smartport_sbus_inverted_uninverted_pin.jpg](url)

sorry @mikeller , I know it is not a configuration troubleshooting thread, RCGroups is for that, but maybe many people are overlooking at this and will prevent them from asking the same all over again, again my apologies in advance.

michael-blue2, 2018-01-25:
@naitse i have tried everything. with univerted and played with all the setting  in bf. Finally i flashed the r-xsr to the latest lbt version. (i am using fcc) and then again i flashed it back to the fcc versio . which i had flashed it over 3 times with. Now everything works. But i do t have the f000 sensor Taranis (qx7)

thooijer, 2018-03-20:
Frsky briefly had a 2018 firmware on their site , so i flashed it because bf script didn't work...telemetry did.
I used the LBT version (have an LBT taranis) of the firmware... nothing worked exept the binding.
Flashed is to ver.171103 LBT and now have it all working. In the meantime the 2018 version i saw on the site has vanished...

jomamamoto, 2018-05-08:
did anyone ever find anymore resolutions to this issue. I have tried all that you have tried, to no avail. My telemetry screen shows RSSI=98db, RXBT=5.0v,A2=---,F000=0 and nothing else. NO PIDS OR VTX HELP

trojanGoat, 2018-05-29:
@jomamamoto,  there's usually a few things that can cause telemetry to not show
1) make sure your wiring is correct and your telemetry wire is going to tx
2) setup the correct uart in betaflight for telemetry (if using f.port just enable serial rx)
3) enable telemetry
4) in some cases you might need to enable / disable telemetry half duplex
5) flash newest rx firmware

zedifpv, 2018-06-04:
I'm having the same issue in 3 quads that was working fine, but when I replaced the RX from X4R to R-XSR simply the FC doesn't detect the VTX anymore!
I already tried every FW at all components and in all quads, but all of them only works with the X4R. The only thing I didn't do yet was to downgrade the FW on the RX or try to flash the XR4R FW on the R-XSR. What do you think about it?

Jyx, 2018-06-04:
One more guy, same issues (only `RSSI`, `RxBt` and `F000`) running the [Emax Hawk 5](https://www.emaxmodel.com/hawk5.html) (Magnum F4 AIO, Omnibus F4 stack), I've tested:
1. All firmware available at the [FrSky page for the R-XSR](https://www.frsky-rc.com/r-xsr/) (mine is EU/LBT, but I've also tried FCC)
2. All firmware available for Taranis Q7XS (tried both EU/LBT and FCC to match what's in the R-XSR for the moment).
3. Used both `TX3` and `TX6` on the flight controller (no traffic seen when running an oscilloscope).
4. `tlm_inverted` ON/OFF, `tlm_halfduplex` ON/OFF in CLI.
5. Both inverted and uninverted (from the pad) looks sane when just checking the data using oscilloscope and without connecting them to the flight controller.
6.  SoftSerial, disabled TX3 B10 when adding a new `resource` in CLI and and tried to use that as SoftSerial (Soft serial enabled on the `Configuration` page.
7. Of course enabled `Telemetry` on the `Configuration` page.

What puzzles me is that I don't see anything on "3" in the list above, I can only see that it goes high (3.3V) when turning on the flightcontroller (and stays at 0V if I disable the `UART`). Shouldn't I see any traffic on the TX pins when I've enabled Telemetry in "Configuration" and in Smart Port for the respective UART? Or is there some handshake protocol between the flightcontroller and the receiver before telemetry data occurs on the TX pins? 

I have not tried `F.Port` and I haven't tried update the flight controller to a more recent version. The Emax Hawk 5 stock configuration comes with Betaflight `3.2.2`.

mikeller, 2018-06-05:
@jyx: Using the uninverted pad on the RX, `tlm_inversion = off`, `tlm_halfduplex = on` should do the trick for your hardware configuration.

Jyx, 2018-06-12:
> @Jyx: Using the uninverted pad on the RX, tlm_inversion = off, tlm_halfduplex = on should do the trick for your hardware configuration.

After everything I tried it turned out to be a rather stupid thing from my side. The UART in question had MSP set in Betaflight and not Configuration. After switching that to Configuration I immediately saw nice traffic on the oscilloscope and indeed, telemetry worked as expected. For someone else trying to debug. I'm running the LBT/EU version `170901` on the R-XSR and I'm using the inverted pad (i.e., I've solder a cable myself). It's connected to `UART6` (`TX`) on the Magnum/Omnibus F4 stack (Emax Hawk 5). I'm pretty sure I'm using the `tlm_...` settings as stated above by @mikeller (if I'm wrong when checking, I'll update this comment). My Q7XS works fine with both the FrSky May 2018 release, as well as `OpenTX 2.2.1` (also EU/LBT). Thanks!