betaflight/betaflight-tx-lua-scripts #45: No Script available in Horus Screens

JoeDaFPV, 2017-10-02:
Hi together,

don't know if I am right here with my question.
Have the betaflight lua running on my horus and it works fine as fare as I climb  through the SD-Card menu and execute the script in the folder.
But I find no way so far to run it on a screen as it is possible with the X9D.

Is there any option for the horus in schedule?

Thanks in advance and best regards,
Joe

raphaelcoeffic, 2017-11-14:
@JoeDaFPV opentx does not allow anything else. There are plans to allow widgets to access key strokes, but this is not the case right now. I will however code some widget that will at least let you see a few key numbers like the VTX settings. See https://github.com/opentx/opentx/issues/3821 for more details.

lshems, 2018-03-30:
I don't know which keys are used to control the lua script. Could anybody clarify, so I can adapt the button simulator?

cromartic, 2019-04-05:
@lshems is there already a way to open the lua scripts in the widget? The keys are the scroll wheel for scrolling up and down, increasing and decreasing value, and pressing the scroll wheel, short press for enter and exit, long press for menue.

lshems, 2019-04-05:
Point me to the script and I'll check.

Send me a pm as well on rcgroups.

cromartic, 2019-04-10:
I mean the betaflight-tx-lua-scripts, especially vtx.lua.
Then I always can see my fpv channel and vtx power.

lshems, 2019-04-10:
I'm not into betaflight, so please point me to the latest and greatest place to find these scripts. Then I can check how to load as a widget. Please do so on rcgroups, as this is off topic here.

cromartic, 2019-04-10:
The betaflight lua scripts are here https://github.com/betaflight/betaflight-tx-lua-scripts/releases

lshems, 2019-04-13:
Ok, First thing, if you want to develop scripts, make sure they run on the simulator. You put in a check that it can't run on the sim. Why? Special reason?

lshems, 2019-04-13:
So I looked at the scripts. On the horus, for widgets, you have to use switches to mimic the buttons. I normally assign two switches, one for enter/exit, and one for plus/min.
On my own gui, I use a page selector screen (menu), where you can go up and down though the menu with plus and minus. Enter will select that page for display. Exit will close that page again.

this gui works with page up and page down. So I would need to ad yet another switch just to use the widget. Not that feasable.

Ask the developer if he could make two selectable fields with mention 'next' and 'previous'. Ask them as well that they will execute a page previous and next command when enter is clicked when that field is selected.

You can then navigate with only enter/exit and plus/minus.

If that's done, you can make a Horus widget pretty easily.

P.S. can you run the script at all on a horus, as one time?

mikeller, 2019-04-14:
@lshems: This sounds like a reasonable option - the horus has lots of independent 'trim' rocker switches, which are pretty much useless for flying quads anyway, so if these could be repurposed to control a widget script that would be great.

Do you want to look into adding this, and then open a pull request for it?

lshems, 2019-04-14:
Ok, I can look at that. I will do some suggestions as well for supporting simulator mode to at least be able to test the GUI.

What is the Horus folder containing now?

mikeller, 2019-04-14:
The script actually supports Horus as it is - the problem is that it cannot run permanently, it has to be started by entering the TX configuration menu, paging to the file explorer page, and manually selecting 'execute' for 'BF.lua'. Making it runnable and usable as a widget script would be a great improvement.

lshems, 2019-04-14:
Ok, I would like to ad a next page field to every screen.

closed 2022-01-17 by klutvott123