betaflight/betaflight-tx-lua-scripts #48: R-XSR 171009, Tramp VTX, Channel 2R and 3R not switchable

m3morx, 2017-10-20:
Hi
Telemetry works fine, but on race band ch 2 and 3 my tramp will not change channel via LUA.
Ch 1 and 5 to 8 works fine. Just switch to 4 from ch 1 is not possible. When you stay at 5 to 8 it works with 4. As well, power switching apply every 5 time you try switching.
During changes, Saving is blinking. Afterwards it refresh and the old value is visible.

Anyone else have the same problem ?
How can o debug the issue ?

Chanelswitching via BF OSD works as desired.
FC = Hobbywing XRotor F4 with the 40A AIO
Telemitry via R-XSR 171009, VTX Immersion RC Tramp HV V2
Taranis Open TX 2.2.0

KIKIFPV, 2017-11-07:
Hi,
I have the same issue.
My config is :
FC = Hobbywing XRotor F4 with the 40A AIO
Telemitry via R-XSR 171009, VTX FuriousFPV Stealth Race
Taranis X9D Open TX 2.2.0
The issue will be due at the Receiver or FC ?!

m3morx, 2017-11-07:
Interesting.
Looks like there is another bug in the R-XSR Firmware or a script incompatibility.
I suggest, the FC could not cause the issue. Because from BF OSD you can control the VTX with no issues. These signals were generated from the FC,
Just interesting, no other user reported any similar issue with the r-xsr.

KIKIFPV, 2017-11-11:
I have sent an email to Frsky with 2 videos. One with a XSR receiver and a second video with R-XSR receiver. With XSR receiver, il is working well.

m3morx, 2017-11-14:
Great! Let me know if you need another video from another User

raphaelcoeffic, 2017-11-14:
as this seems to be an issue with R-XSR, I'll just close this one. Please re-open if it appears to be related to the scripts.

closed 2017-11-14 by raphaelcoeffic