betaflight/betaflight-tx-lua-scripts #53: No Time over MSP support for X7?

Arakon, 2017-11-11:
I installed the current set of scripts on my X7 and latest 3.3.0 master.
Time is apparently not being synced over smartport (R-XSR), but i.e. PID settings work.
Time doesn't show up in OSD when set up to i.e. replace my warning position, and BB logs are not timestamped in any way either. Or am I missing some crucial step here?

mikeller, 2017-11-11:
@Arakon: It _should_ work. Probably the most basic way to verify is to go into the CLI, and see if `status` is showing the time.

I can't verify for X7, it works on X9. Also, probably worth verifying that you are on the latest stable OpenTx 2.2.

Arakon, 2017-11-11:
# status
System Uptime: 12 seconds
Current Time: 0000-01-01T00:00:00.000+00:00

Doesn't look like it.

mikeller, 2017-11-11:
Hmm, now I also recall seeing some discussion around the R-XSR firmware having issues with MSP over SmartPort with the latest firmware. You would not have an XSR handy to test this theory?

Arakon, 2017-11-11:
Unfortunately, not set up. However, the R-XSR didn't have smartport PID support UNTIL the latest firmware, they fixed those issues.. and the PIDs are working fine.

Arakon, 2017-11-12:
I just set it up on a spare Omnibus F3 I have, but it seems the time feature isn't there on F3 at all? I'm not even getting the time line in Status in the CLI.

mikeller, 2017-11-12:
Erg. It had to be removed from the OMNIBUS F3, since there wasn't enough flash space. It is enabled on other F3 boards.

Arakon, 2017-11-12:
Strange. Works on the XSR on F3 apparently, I just kicked out baro support to test.
Doubly strange.. it works on R-XSR now too. Could it have to do with the order of switching on devices? I also updated firmware again to today's master at the same time unfortunately.

I did get several occassions where the time would not show up now, but after turning both devices on and off a few times, it showed up again.

mikeller, 2017-11-12:
There is also #54 that should fix the problem that time was lost when the flight controller rebooted, but the RX was not power cycled. Maybe that is what is giving you inconsistent results.

Arakon, 2017-11-12:
Ah! That sounds like it could well be the issue.

Arakon, 2017-11-12:
Appears to have fixed it or at least made it more consistent.

closed 2017-11-12 by Arakon