betaflight/betaflight-tx-lua-scripts #55: Will you merge with codecae script sometime?

RipperDrone, 2017-11-13:
There is a mighty, extended functions script available here - will you merge it?

https://github.com/codecae/betaflight-tx-lua-revised

mikeller, 2017-11-13:
Well, merging it will not work in a simple way, since https://github.com/codecae/betaflight-tx-lua-revised was not created as a fork of this repository. :confused: 

There are some open points with https://github.com/codecae/betaflight-tx-lua-revised that I would like to see addressed:
- switching to a build system that does not rely on CI processes to have write access to the repository, and push changes into the repository, but instead use the 'release' functionality of GitHub to package releases and make them available to users;
- adding support for (at least) Horus and X7, so that existing users of this repository who own these devices are not left out.

Once these have been addressed, we are going to look into doing a pull request to replace this repository with the contents of https://github.com/codecae/betaflight-tx-lua-revised (provided that @codecae is on board with this).

codecae, 2017-11-13:
@mikeller Totally on board for sharing this code!

Regarding your points:
Point 1 -- Working on this as we speak.
Point 2 -- There are X7 and Horus branches underway.  They are not current merged due to differences in the Crossfire scripts, but I'm beginning to wonder if they should be part of this mix anyway.

Once I get the branches cleaned up and pushed, we can start laying out an action plan.

mikeller, 2017-11-13:
Nice!

See codecae/betaflight-tx-lua-revised#62 for some pointers on how to tackle 1. (I think if we switch to precompiling the scripts then the need to minify them will go away, since they don't have to be compiled on load any more.)

Re 2, I would like to help, but I don't own a Horus or X7, so I am limited with respect to these. But in general, I think if it is feasible to at least have the SmartPort and Crossfire scripts built on top of the same libraries (i.e. common things like UI, MSP), then that should be the way to go, even if we move to building them in separate 'packages', since most users won't be using both protocols concurrently.

codecae, 2017-11-13:
I've removed the minification scripts and the sdcard directory, so the heat is on to get the release zips published.   I think I'll go ahead and tag a v1.0 and manually zip up the initial set of minified scripts.  Then I can update the readme to link people to those zip files for downloading purposes.

Once that's complete, I'll finish up the Jenkins jobs to build all three branches for now (X9, X7, Horus) and publish build artifacts for download there.  I'm about 50% done with the pull request jobs so testers can use minified or compiled code to dodge any unwanted memory issues.

I'll try to wrap it up in a few days.

codecae, 2017-11-13:
Re 2: There are common libraries that are already built in the revised scripts.  The screen code and associated logic is built into separate "packages" persay, that all apply to the same common set of libraries.  When it comes to SmartPort and Crossfire, the scripts determine which telemetry protocol to use by pinging each of the telemetry buffers.  Some people (including myself) have models on SmartPort and Crossfire, so having this seamlessly built into the scripts is pretty excellent.  No reason I can think of to bifurcate stuff into different releases.  Eventually, all models and protocols can be merged into a common master and nobody will need to care which protocol or radio they are using.

mikeller, 2017-11-13:
Ok, looks like I misunderstood your original point then. What did you mean should not be part of this mix?

codecae, 2017-11-13:
Sorry -- for clarification, I was referring to the CROSSFIRE scripts.  To me, this should be on OpenTX.

mikeller, 2017-11-13:
Ah. To some extent this can be said for the SmartPort script as well. Ideally, they should have something like a 'marketplace' for people to host their script offerings. But way back when, in the absence of that, we decided to set up a repo for the scripts in the betaflight project, so that collaboration could happen. :-)

codecae, 2017-11-13:
We could continue to offer them, but they will need to be modified to be dynamically loaded.  There are differences in the LCD screens (sizes and APIs) that prevent a merge at this time.

mikeller, 2017-11-13:
> We could continue to offer them, but they will need to be modified to be dynamically loaded. There are differences in the LCD screens (sizes and APIs) and controls that prevent a merge at this time.

The builds for the different devices? We could always offer them in the form of one build for each device.

codecae, 2017-11-14:
The code has been rebased on the former betalight-tx-lua-revised...closing!

closed 2017-11-14 by codecae