betaflight/betaflight-tx-lua-scripts #82: Horus X10,X10s support

gumush, 2017-11-15:
Not working in nightly build opentx in X10s.

raphaelcoeffic, 2017-11-15:
@gumush could you please be more precise? I'm using myself a X10 and it works like a charm.

gumush, 2017-11-15:
I'll check again and post the error message.

raphaelcoeffic, 2017-11-15:
@gumush ..and your exact opentx version, and everything else. Did you actually make sure that SPORT is working properly? Did you try it on another TX, like the X9D, or whatever you might have?

codecae, 2017-11-16:
@raphaelcoeffic might be worth updating the readme to include instructions for Horus and X7 at some point.

putimir, 2017-11-16:
Hi, I'm also getting errors now, on all systems that were previously working with the "revised branch"...both crossfire and smartport

I hadn't touched except loading the new scripts...

I'm using Otx 433, Horus X10S, and various omnibus boards (OBF4,OBSD), BF 3.2.1 and 3.2.2. I am getting the telemetry and if I revert back to scripts (without "background separation" - or what did you call it) all is well, so I can be sure, all my systems are properly configured.

I AM running, as I always did /SCRIPTS/TELEMETRY/bf.lua

With the new release, I'm getting a dim blank screen with no error; if I run the script in the sumulator, I get at the end:

```f_open(D:/RC/FrSky/SD/Contents/SCRIPTS/BF/HORUS/pids.luac, 1) = 0A5C8A28 (FIL 0B77EC20)
f_close(0A5C8A28) (FIL:0B77EC20)
Script run function returned unexpected value
luaInit
luaClose 15BB03D0
lsScripts 15BB13C0```

putimir, 2017-11-16:
Also, strangely, sometimes I get the error:

```BF/MSP/common.lua:104
attempt to perform artihmetic on global'.....```

raphaelcoeffic, 2017-11-16:
@putimir please run `/BF/bf.lua`, I will update the instructions for the Horus TXs ASAP.

putimir, 2017-11-16:
Maaan! I'm sorry, how did I miss this one?? I was focused on the contents of the zip file, which does not contain the mentioned script. :/ 

I'm sorry, thanks!

codecae, 2017-11-17:
I will update the Readme to contain a proper link.  It only contains the scripts folder.

codecae, 2017-11-17:
For now, please use [this link](http://null.eastus.cloudapp.azure.com/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+/lastSuccessfulBuild/artifact/obj/*zip*/obj.zip).

sniperxnl, 2017-11-30:
I have the sam issue. I unzip the folders BF and Script to my sd card.
I go to BF folder and i mis the BF.lua, i see this in file name in scripts, but when i excute that nothing happend.

s700ky, 2018-02-06:
I have managed to get mine work by this:
Bf.Lua, KissCF.lua, KissSP.lua onto the card root file.
BF folder into the scrips folder, and Bf.Lua file in Telemetry folder.
Kiss folder and KissCF.lua and KissSP.lua into the Telemetry folder.
Execute BF or Kiss from the root file and they run.
However, I cannot work out how to make them appear as a Telemetry screen or be able to select them on the LUA screen.

codecae, 2018-02-06:
You should not be required to extract files in a way other than how they are packaged.  Horus does not support telemetry scripts, so simply execute the /BF/bf.lua file instead of the telemetry script.

s700ky, 2018-02-06:
I tried that and I got an error message. 
The way I have done it works?

closed 2018-05-13 by stale[bot]

lswright, 2019-05-02:
Has anyone solved this.  Same problem here.  I've tried running the bf.lua from /BF folder on SD card and also from SD card root.  Still just a flash of PIDs and then a slightly dimmed screen.  Any solutions?

Cheers

> Hi, I'm also getting errors now, on all systems that were previously working with the "revised branch"...both crossfire and smartport
> 
> I hadn't touched except loading the new scripts...
> 
> I'm using Otx 433, Horus X10S, and various omnibus boards (OBF4,OBSD), BF 3.2.1 and 3.2.2. I am getting the telemetry and if I revert back to scripts (without "background separation" - or what did you call it) all is well, so I can be sure, all my systems are properly configured.
> 
> I AM running, as I always did /SCRIPTS/TELEMETRY/bf.lua
> 
> With the new release, I'm getting a dim blank screen with no error, more precise the pids screen brielfy flashes into view and then dissapears into blank screen; if I run the script in the simulator, I get at the end:
> 
> ```
> f_close(0A5C8A28) (FIL:0B77EC20)
> Script run function returned unexpected value
> luaInit
> luaClose 15BB03D0
> lsScripts 15BB13C0```
> ```

mikeller, 2019-05-04:
@lswright: There is a problem with the version detection in the current release. Can you please try this: 
[betaflight-tx-lua-scripts_1.2.1.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/3144024/betaflight-tx-lua-scripts_1.2.1.zip)

lswright, 2019-05-05:
Thank you @mikeller.  I haven't had a chance to try this yet but will do as soon as I can.

Regards

Lesa