betaflight/betaflight-tx-lua-scripts #89: RSSI over MSP not working?

Arakon, 2017-11-18:
Maybe I misunderstood the feature. X7, latest script release (1.0). I disabled RSSI analog and channel. RSSI on OSD now remains stuck at 99. Am I mistaken in thinking that the FC should be getting RSSI from the smartport link? Everything else works, i.e. PID settings etc.

mikeller, 2017-11-18:
No, that's how it should work (i.e. all other sources for RSSI have to be disabled). Can you get latest on 3.3 and try again? There have been some changes to the 'RSSI over MSP' implementation. N.b. that RSSI will only be sent by the TX when it is _not_ currently showing the Betaflight lua screen. This also gives you an easy way to test if it's working: After entering the lua screen, RSSI should dropt to 0 after 1.5 seconds, and come back after leaving the screen.

Arakon, 2017-11-18:
Just tried with the current Master of 3.3, It shows a flashing 0 RSSI now with RSSI ADC and channel disabled.

mikeller, 2017-11-19:
That happens even when you are *not* on the Betaflight lua screen (i.e. 'normal' telemetry values showing)?

Arakon, 2017-11-19:
I just updated FW again and it appears to work now, not sure if this was related to not power cycling properly between saving settings or something. However, the displayed RSSI value in OSD and the value shown on the transmitter differ pretty badly.. OSD shows 99-97 while the transmitter shows 89-86.

mikeller, 2017-11-19:
@Arakon: Transmitter is dBm, OSD is scaled to percentage (also, a guesstimate of what the percentage should be so far.

Arakon, 2017-11-19:
Ah, alright. Although, wouldn't it make more sense to show the actual value from the transmitter, since it's available in "true" form?

mikeller, 2017-11-19:
The RSSI value is defined to be 0 to 100%. Who knows how other systems (like MWOSD) will react if it suddenly reports 120?

KnuckleUpFPV, 2018-01-04:
If it reports 120 the osd reads 99 until the rssi drops to below 99 on the tx. Then its one for one all the way down to failsafe. This is true with betaflight osd, mwosd, and mateks hubosd (which i believe is based off mwosd). So if transmitter reports 92 the osd reports 92. There is no delay in osd. When tx rssi hits 100 again the on screen osd goes back to 99 and stays there until the link quality drops again. This is how it works with openlrs. Rssi on tx starts around 121-124, while osd reports 99. Using a 30db attenuator I walked away from the receiver with tx and goggles. Rssi drops all the way 0 before failsafe.

seaweeduk, 2018-03-12:
I tested RSSI over MSP last weekend on two of my builds, both Betaflight F3 with R-XSR's, no soft serial is used. My transmitter is an x9d+ Taranis with the latest version of the scripts and opentx. 

Quad 1 :
First flight: around 2/3 through the first battery the RSSI stopped working and dropped to 0 and stayed there, no telemetry lost or RSSI low warnings were issued by my TX. Once I landed and tried opening the lua scripts menu, the menu never retrieved my PID values. I power cycled the transmitter, with the quad still powered and the RSSI reading returned to the corner and the lua scripts started functioning again. I closed the lua menu and flew the entire second battery without RSSI display problems.

Quad 2:
I powered this quad up without power cycling my transmitter from the first quad, the RSSI reading stayed static at 98 for the entire flight without moving. I didn't get a chance to do a second flight on quad 2 before the rain stopped me flying. 

I have log files and DVR of both quads available if they are useful for debugging the problem.

closed 2018-05-13 by stale[bot]