betaflight/betaflight-tx-lua-scripts #90: Syntax error background.lua line 12

RadianceNL, 2017-11-18:
With the new 1.0 lua scripts I get a syntax error in backgrounds.lua on line 12, attempt to index a nil vallue.

I'm own a X7 and the previous version 0.3 is working just fine.

Any ideas?

mikeller, 2017-11-18:
@codecae: Looks like a guard is needed for the case that `getFieldInfo(protocol.stateSensor)` returns `nil`.

codecae, 2017-11-18:
It is hardcoded to both protocols.  Not sure why this would occur.  Perhaps @RadianceNL has an older version of protocols.lua.

mikeller, 2017-11-18:
I was more thinking of the `getFieldInfo()` call returning `nil` for whatever reason. But it could be caused by an older version of OpenTx. Is there a way to detect the OpenTx version in lua? If so, we should probably add a warning for users trying to run on an old version.

codecae, 2017-11-18:
Perhaps a getVersion() fast fail as part of radios.lua is in order.

codecae, 2017-11-18:
Here
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/radios.lua#L38

RadianceNL, 2017-11-18:
Forgot to mention that I am running OpenTX 2.2.0 and as far as I know that is the latest version?

codecae, 2017-11-18:
It should work fine with 2.2 .. can you try deleting the scripts from you
SD and recopy them?

RadianceNL, 2017-11-18:
Sure I will but I've already done that actually this afternoon. I will redownload the scripts and try again.

codecae, 2017-11-18:
Try rediscovering sensors

codecae, 2017-11-18:
Is it possible that you're using BF.lua instead of bf.lua?

RadianceNL, 2017-11-18:
Already done that as well, delete all sensors and then discover them again but didn't do the trick.

codecae, 2017-11-18:
@mikeller it looks like in the v1.0 zip that the return is on line 12.  This should be fixed in master, right?

RadianceNL, 2017-11-18:
When I select the script I see bf, so no capital letters like the previous version had.

On my SD card I have bf.lua as well, not BF.lua

codecae, 2017-11-18:
Do you have a Tmp1 sensor in your list of telemetry sensors?

RadianceNL, 2017-11-18:
Yes, it's reading 10011 to 30011 degrees celsius.

When I didn't had telementry I was able to start up the lua script but once telementry got recovered I get syntax error BF/MSP/common.lua line 107.

Attempt to perform arit.....

RadianceNL, 2017-11-18:
And after another reboot of the transmitter everything started to work...very strange but thanks a lot for the help.

Although I'm curious what happend

mikeller, 2017-11-18:
@codecae:

>  it looks like in the v1.0 zip that the return is on line 12. This should be fixed in master, right?

Looking at the zip, `background.lua` is:

```
mikeller@pc:~/tmp$ cat -n obj/SCRIPTS/BF/background.lua | head -20
     1	local INTERVAL          = 50         -- in 1/100th seconds
     2	local MSP_SET_RTC       = 246
     3	local MSP_TX_INFO       = 186
     4	
     5	local lastRunTS
     6	local sensorId = -1
     7	local timeIsSet = false
     8	local mspMsgQueued = false
     9	
    10	local function getSensorValue()
    11	    if sensorId == -1 then
    12	        sensorId = getFieldInfo(protocol.stateSensor)['id'] or -1
    13	    end
    14	    return getValue(sensorId)
    15	end
    16	
    17	local function modelActive(sensorValue)
    18	    return type(sensorValue) == "number" and sensorValue > 0
    19	end
    20	
```

To me it looks like there is something dodgy with sensor discovery, and that causes `getFieldInfo()` to return `nil` in some cases.

My hunch would be to leave it as it is for now (or do a quick fix fending against the `nil` return), and then rewrite the whole polling once betaflight/betaflight#4611 is in, and make it use MSP to query the status of RTC and RSSI, and act accordingly.

codecae, 2017-11-18:
Miscounted the lines on my phone.. my bad :)

knockoutperformance, 2017-11-20:
I'm getting nil return on line 12 also on a qx-7 and r-xsr with version 1. Running opentx2.2 newest release, internal module updated and the r-xsr. I have to turn off receiver to recover telemetry screen. But when it does work, it will change pids.

robschii, 2017-11-20:
i have to choose rxBt @ display -> infoline : voltagesource

codecae, 2017-11-20:
Please try this version of the scripts to see if it overcomes your sensor discovery issue (line 12 error)

http://null.eastus.cloudapp.azure.com/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/24/artifact/

knockoutperformance, 2017-11-20:
I tried it for a few flights. No more sensor issue(would get error on line 12 even before flying). Thank you. Very awesome to be able to change pids from the controller. I don't use vtx yet, that will come in time.

codecae, 2017-11-21:
@alejoal07 @RadianceNL can you also please confirm whether or not the scripts provided above correct the issue you're experiencing?

toddsalpen, 2017-11-21:
I have the same error and i am using a Taranis X9D+SE with OpentTX 2.2.0
codecae i will try your new files

codecae, 2017-11-22:
This issue should now be addressed in release [v1.0.1](https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/v1.0.1).

Please download and verify!

alejoal07, 2017-11-22:
Tried v1.0.1 and it works now, but values don't load for pages
- Rates 2/2
- Filters 

Haven't tried modifying stuff from within the Script yet, but I don't get any values loaded on the script for those pages. 

Betaflight I'm running is 3.1.7

codecae, 2017-11-22:
It's recommended to use the most recent stable release.

alejoal07, 2017-11-22:
Got it,
I will set up 3.2.2 on my new build and let you know how it goes.
Also, when I select the fields that are not loaded and try to place values, I get:

Script syntax error
BF/ui.lua:231
Attempt to perform arithmetic on field 'val

While playing around with the script I did also manage to get a "Not enough memory" error but I' my not sure how to replicate that.
My radio is an X9D+ running OpenTx 2.2

codecae, 2017-11-22:
It's likely you'll encounter these issues if you're using an older version of Betaflight.

RadianceNL, 2017-11-23:
Didn't had the time to test it yet, will come back at it this weekend. Thanks for the quick support!

RadianceNL, 2017-12-04:
Forgot to report back but I can confirm that the issue is solved. I don't get any errors anymore and settings coming up as intended.

closed 2017-12-04 by codecae