betaflight/betaflight-tx-lua-scripts #93: Request to change order of pages.

kmitchel, 2017-11-20:
Please consider making the vtx page first, followed by rates, then Pids.  Seems silly that I should have to page thru everything to get to the one page I am most likely to use. Ty

raphaelcoeffic, 2017-11-20:
You're probably right: that's basically the page order ranked by usage count.

kmitchel, 2017-11-20:
I'll need to check the code tonight and see if I can reorder easily--for my
own sanity.  I am very grateful to be able to tune dterm set point and gyro
filters.

codecae, 2017-11-20:
It's simple to reorder.  Just need to update the 'preload' scripts for each of the templates.  The PageFiles table contains the template pages in the order they are displayed.

mikeller, 2017-11-20:
@codecae: In general, but especially if we reorder the pages so that the VTX page is shown first, we should change it so that the VTX page is shown only if a configurable VTX is set up - it's frustrating for a user to have to click through a nonfunctional page for hardware he does not have every time.

knockoutperformance, 2017-11-20:
@codeac: same as mikeller, maybe have vtx page if using hardware. I don't use vtx right now, but eventually will.

kmitchel, 2017-11-20:
Sweet...just was getting ready to start...distracted by fport!

closed 2017-11-20 by kmitchel