betaflight/betaflight-tx-lua-scripts #95: Taranis Q X7S not supported

jakezilla, 2017-11-21:
The Taranis Q X7S seems to have a version of firmware that isn't mentioned anywhere on the internet. (if anyone has any info on where to actually find this, please let me know)

FW: opentx-x7s
VERS: 2.2.0 (9ad5a5be)
DATE: 2017-09-04 20:37:35
EEPR: 218

Anyway, it seems to report its radio type as "x7s" and not "x7" and the BF Lua reports it as an unsupported radio.

Inserting this line in the function supportedRadios in the file radios.lua
`["x7s"] = supportedPlatforms.x7,`
Makes it work 'properly', although then I get the Syntax Error already logged with issue #90 

I've never made a contribution to a GitHub, and I'm not sure on the process, otherwise I'd just add it.

mikeller, 2017-11-21:
Damn, you're too fast, @codecae. I was going to encourage @jakezilla to learn how to GitHub today:

https://gist.github.com/Chaser324/ce0505fbed06b947d962

codecae, 2017-11-21:
hah I already had it open.  I can cancel if you want.

closed 2017-11-21 by jakezilla

mikeller, 2017-11-21:
@jakezilla: No problem, thanks for helping us with feedback and reviewing. Community driven projects like Betaflight can only turn out great things because a lot of people put in effort to make them better.