betaflight/betaflight-tx-lua-scripts #97: Error: Not enough memory on 2.2.1N439 opentx nightly

brycedjohnson, 2017-11-21:
Was testing out the latest opentx nightly for a different reason and I saw I was getting a memory error.  Seems to happen after paging through a few lua screens.

Script Syntax Error
BF/ui.lua:335
not enough memory

codecae, 2017-11-21:
@brycedjohnson did you download the source or use the release zip file?

brycedjohnson, 2017-11-21:
@codecae Release zip file.  Should I try the source?

codecae, 2017-11-21:
no the release is the smallest in terms of memory.  Which page is loading when you hit the limit?

brycedjohnson, 2017-11-21:
On the 4th or 5th page in.  Weirdly one time it just worked and I didn't have any issues until I rebooted the taranis.

codecae, 2017-11-21:
Would you mind trying this zip? 

http://null.eastus.cloudapp.azure.com/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/24/artifact/obj/*zip*/obj.zip

brycedjohnson, 2017-11-21:
That seemed to fix it!  Tried it like 3 times and it worked everytime.  Thanks!

brycedjohnson, 2017-11-21:
Want me to close this issue or want to close it when you reference whatever fix you did?  Thanks for the quick help!

codecae, 2017-11-21:
I'd like to leave it open for tracking purposes.

Glad it helped you :)

@mikeller Looks like the Jenkins build didn't have the -s switch for luac.  Should we do an increment release with the current master and include a better build, or should I just update the release zip to contain a build using -s?

mikeller, 2017-11-21:
@codecae: No problem, we are still in the process of getting this set up. I would say there have been a couple of pretty significant bug fixes since 1.0, so a release 1.0.1 would be in order. (Also, we should totally move to SemVer (https://semver.org/).)

One of the things in SemVer that is sometimes overlooked, but in my opinion is a crucial part of it is 

```
Once a versioned package has been released, the contents of that version MUST NOT be modified. Any modifications MUST be released as a new version.
```

Breaking this rule will create a lot of confusion and frustration amongst users ('I tried v1.0, it did not work, why would I try it again') for no benefit at all (there is an unlimited supply of version numbers, and they are free ;-) ).

David-VG, 2017-11-21:
I am not sure if it's the same bug or not, but I am using OpenTX 2.2.0 on a Taranis X9D:

Script syntax error
TELEMETRY/bf.lua:10
not enough memory

I get this with the release 1.1 and the file http://null.eastus.cloudapp.azure.com/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/24/artifact/obj/*zip*/obj.zip

David-VG, 2017-11-21:
Also disconected but switching between pages I get:

Script syntax error
BF/ui.lua 335
not enought memory

David-VG, 2017-11-21:
As a little tip: I am using 2 telemetry scripts, one is the betaflight tx lua scripts and the other one: https://github.com/tozes/taranis_telemetry

I am not sure, but maybe betaflight tx lua scripts needs ALL of the available memory.

raphaelcoeffic, 2017-11-21:
@David-VG yes, definitely! but you could use the one-time script (in `/BF/bf.lua`) for configuring, and the other as a telemetry script for flying.

raphaelcoeffic, 2017-11-21:
Or just my yourself a brand new shinny X10: no issue with memory there ;-)

David-VG, 2017-11-21:
I dont get Memory errors that way, (running BF.lua manually). But just wanting to warn about the memory issues. Thanks @raphaelcoeffic

mikeller, 2017-11-21:
@raphaelcoeffic: Except that the X10 can not (yet) run telemetry scripts at all (as far as I know), and the https://github.com/tozes/taranis_telemetry does not yet support the X10. ;-P

raphaelcoeffic, 2017-11-21:
@mikeller yeah, it's called "widgets", and these scripts will need to be modified. And it still lacks the capability of having widgets that can handle keys. But we're working on it, you can bet on that ;-)

mikeller, 2017-11-21:
@raphaelcoeffic: Just hassling you. :-P You are involved in OpenTx? Also, out of interest, what are the MCU / RAM hardware specs of the X10?

raphaelcoeffic, 2017-11-21:
@mikeller I did only a couple of small things. Trying to get more involved these days, so I can make my new X10 with opentx the best TX in the world :-D

mikeller, 2017-11-21:
That's a good goal. Having nice new hardware but firmware that isn't as good as the one on the old stuff is always a big disappointment.

I've seen flashy marketing videos of FrSky's own firmware on the X10. Is it any good, or is it just a non-recent version of OpenTx with some flashy bitmaps slapped on?

raphaelcoeffic, 2017-11-21:
Totally different. It's quite good for people in need of a UI that basically works like on of the main brands (futaba was named a few times in that context). But it's not my cup of tea when it comes to planes. I would not mind for Quads. probably enough (no LUA though).

codecae, 2017-11-22:
This issue should now be addressed in release [v1.0.1](https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/v1.0.1).

Please download and verify!

David-VG, 2017-11-22:
Still out of memory here :(

But thanks.

codecae, 2017-11-22:
I'm wondering now if the other script you're using is eating up resources.    These scripts have been dialed back and optimized in just about every way possible to reduce the memory footprint.  I'm not sure if the other script you're using has undergone the same level of scrutiny.

codecae, 2017-11-22:
Have you tried temporarily disabling the other script and seeing if it still happens?

David-VG, 2017-11-22:
It works while using just of both scripts (even if I dont power cycle).

Also works if y manually launch BF/bf.lua while the other script is assigned as the only telemetry script.

But the other script only runs while being displayed (does not background function, just "run").

https://github.com/tozes/taranis_telemetry/blob/master/X9D/TELEMETRY/telem1.lua

As you can see is just a run, divided in 9 functions that just draw the screen. I will try other script just to know if the problem is having 2 telementry scripts or just that "telem1.lua".

raphaelcoeffic, 2017-11-22:
> But the other script only runs while being displayed (does not background function, just "run").

That does not matter at all: it is still kept in memory, as long as you model is loaded and no one-time script is executed.

David-VG, 2017-11-22:
Yeah, but it doesent use global variables, just the run loop that just draws.

raphaelcoeffic, 2017-11-22:
The code is stored in memory as well

David-VG, 2017-11-22:
Yeah, but only 238 lines. I tought the bitmaps could be the issue but without bitmaps had the same problem.

Tried a hello world without issues and also the snake.lua. Will keep trying.

codecae, 2017-11-23:
@mikeller how do you feel about the suggestion of dynamic loading of the background script?  ie: setting the background object to nil when the telem screen is loaded and dynamically reload it when the telem screen is exited

mikeller, 2017-11-23:
@codecae: If that helps with the memory problem (i.e. the entire script is kept on the heap, and not just function pointers), then that's definitely a good idea.

brycedjohnson, 2017-12-09:
I was getting some memory errors again.  Might have been after updating to the 1.0.1 release rather then the zip you included on here.

I just updated to 2.2.1RC1 opentx and have lau and lauc (not sure if that is needed) for options. (opentx-x9d+-ppmus-multimodule-noheli-lua-luac-en.bin)

brycedjohnson, 2017-12-09:
Well now it is working and I can't cause the memory error.....

codecae, 2017-12-09:
I would think that nightly firmware releases are dodgy.  Any reason why you don't want to just stick with the 2.2 stable release?

brycedjohnson, 2017-12-09:
Testing out some multimodule features that were added.  The last test was with the 2.2.1 RC, so any issues that come up are likely to make it into 2.2.1 if they don't fixed.  I'll likely put 2.2.0 back on there, but I thought I'd update the issue.

nikybiasion, 2017-12-21:
With opentx 2.2.1 rc3, i let the radio to compile the luac files and then i 've deleted the lua files. This is the only way to avoid memory messages. 
I flash the firmware with lua and luac support

codecae, 2017-12-21:
@nikybiasion have you tried only copying over the .luac files?  Also -- they have 2.2.1 in general availability.  rc3 may have some bugs that were addressed in the general release.  can you try with the 2.2.1 general release?

nikybiasion, 2017-12-21:
I've only tried to build luac on my pc and then copy on the radio, but may be the binary format was different and doesn't work, i can try to clean sdcard and copy only luac. I will try the official 2.2.1

codecae, 2017-12-21:
you shouldn't need to build the luac files yourself.  They're included with the release.

codecae, 2017-12-21:
if you choose to build them yourself, make sure to use the -s switch to strip the debugging information from the binaries.

Ga1j1n, 2017-12-24:
I've tested with 2.2.1 release version and the memory issue remains, I had to downgrade which is a shame

codecae, 2017-12-24:
Which screen causes the memory errors to manifest?

Ga1j1n, 2017-12-24:
Sorry, I've downgraded now but I'd say you get 3-4 screens in before the error occurs

stoms12, 2017-12-26:
I am also getting out of memory error. Running Lua script v1.0.1 and OpenTX 2.2.1
I seem to get error either running script manually or as a telemetry screen.
Any progress on fix?

codecae, 2017-12-26:
Which screen is causing the error?

stoms12, 2017-12-26:
Can’t even load the script. I am also running https://github.com/tozes/taranis_telemetry

nikybiasion, 2017-12-26:
I've flashed 2.2.1 opentx release (lua+luac)
I've reinstalled the script, just after entering in the first screen, i can see only 336b free memory...
If i copy only the luac from release 1.0.1 zip file, i get 'bf/x9/rates1.luac incompatible precompiled chunk'
If i copy only lua, all works fine

codecae, 2017-12-26:
You shouldn't need the lua and luac options unless you plan on using special functions.  Can you try without for testing purposes?

nikybiasion, 2017-12-26:
Without lua and luac, the radio start with only 144b of free memory, after ciclyng some times in the telemtetry screens, the out of memory message appear

stoms12, 2017-12-26:
I have re flashed firmware and reinstalled script leaving out the telemetry script.
I am able to open it and change pages, exit out, open again and change pages until I get to rates 2 page. Then I get memory error, this doesn't go away until I restart TX.
Seems like everytime the script opens, it opens a new instance and memory is not cleared.

codecae, 2017-12-26:
The script is loaded with the model...  It is not reopened every time you
start it.

codecae, 2017-12-26:
The problem is that the new version of open tx uses more memory that the previous.  This is a very limited and precious resource on these transmitters.

stoms12, 2017-12-26:
Any ideas on what is causing memory issue then?

stoms12, 2017-12-26:
So I guess I could try 2.2.0?

codecae, 2017-12-26:
It is tested and verified on 2.2.0.  this, however, will not guarantee that other telemetry scripts you have running will not get in the way.  They all share the little memory that is available on the hardware.  We have gone to great lengths to ensure ours uses it as efficiently as possible in it's own right.  If another script is running at the same time creating contention, that is out of our control.

codecae, 2017-12-26:
I'm looking to see if there is any way to squeeze it more.

mikeller, 2017-12-26:
Looks like it's time to bring back minification.

codecae, 2017-12-26:
Everything seems to be localized quite well.  Only thing that doesn't appear to be is the values table in the background script.  Its usage size should be fairly benign.  I wonder if something in opentx changed with respect to lua vs luac or garbage collection.

mikeller, 2017-12-26:
@codecae: I meant the programmatical minification, like what you used before we moved to precompiling the `.luac`. Not sure if it gains us anything when done in addition to the precompilation, but it can't hurt trying.

[99 more replies]

closed 2018-05-29 by stale[bot]