betaflight/betaflight-tx-lua-scripts #98: Vtx channel/band/power problem

Brisingrox, 2017-11-23:
With latest v1.0.1 when you try to change vtx settings there is always a retrying message who pop up and even sometimes the script choses random channels. I use the Furious Fpv Stealth Race vtx and a Qx7. I'm not the only one with this issue. Besides this the script works flawlessly on every other page and settings.

codecae, 2017-11-23:
Oddly, nothing has changed in v1.0.1 that would affect this.

Which version of Betaflight are you using?  Smartport? Crossfire?

Brisingrox, 2017-11-23:
Betaflight 3.2.2 and smartport

ethomas997, 2017-11-23:
I also see the 'Retrying' message more often then not after doing a save-page.  I think it's a long-standing issue that has always been there, but was not so noticeable until the 'Retrying' message was implemented.

From what I've seen, the MSP 'set' and EEPROM_SAVE messages almost always go through OK, but very often the MSP response from EEPROM_SAVE message is never received.  If I modify the script so it doesn't do the EEPROM_SAVE then I almost never see the 'Retrying' message.

My test setup is a Q X7, XSR receiver (latest 170619 firmware), and Betaflight master 20171118_build372.  I also get similar results using a FuriousFPV SPD15 receiver.  I guess the issue could be Q X7 related, but I doubt it.

I'll try and do more checking, testing, etc.  If others are seeing similar issues, please chime in...

Brisingrox, 2017-11-23:
It's kind of funny that you say that you use the Furious Fpv Spd15 receiver because that's exactly what receiver I use.

Brisingrox, 2017-11-23:
@etthomas997

How can I modify the script they way you have done it? Seems like a little workaround 
Thanks

ethomas997, 2017-11-23:
@Brisingrox:  It's only really for testing.  If you disable 'eepromWrite' then any changes you make will not be persistent and will revert when the flight controller reboots.

Brisingrox, 2017-11-23:
Ok. Thanks for the information. Looking forward to know what could be the solution for solving this.

codecae, 2017-11-23:
In 3.2.2 the EEPROM write does nothing for the vtx, but likewise, should
have no effect.  This was not introduced in v1.0.1.

codecae, 2017-11-23:
Another thing about 3.2.2 and previous -- settings are applied directly to the vtx, so if they aren't working, it's likely something to do with the connection or configuration.  With which versions did this work?  (Script and BF)

Brisingrox, 2017-11-23:
I used the lua script ver 0.3 with every Betaflight 3.2 version (pre release and rc) without any issues

codecae, 2017-11-23:
Does the vtx channel change, and not the values on the screen?  Wondering if it's being impatient and is failing prematurely.

codecae, 2017-11-23:
Would you mind testing v1.0.1 with the following build of Betaflight using your setup?

https://ci.betaflight.tech/job/Betaflight/348/

Brisingrox, 2017-11-23:
Sure thing. If I could help improve it would be a pleasure for me. I will flash this build and let you know directly

codecae, 2017-11-23:
There are changes that persist settings on the FC, which should get around the potential time out issue.  A little heads up -- it will default your vtx to F1, 25mw the first time you boot the FC.  This is normal.

Brisingrox, 2017-11-23:
Ok thanks for the information will see this in a few minutes. Just backing everything up now.

ethomas997, 2017-11-23:
Version 0.3 of the scripts did not have the "Retrying" message and were configured to do 0 (or maybe 1) retry.  My theory is that these problems were happening then but were going unnoticed.  Please see my previous comment where I describe what I see happening https://github.com/betaflight/betaflight-tx-lua-scripts/issues/98#issuecomment-346512775

Brisingrox, 2017-11-23:
Yes I have read trough your post and understand why this is happening now and not before. Thanks for the information as I already told you.

syrekszymon, 2018-02-03:
Please add correct VTX power level for FuriousFPV Stealth Longrange.

codecae, 2018-02-03:
There is no means to determine which hardware is in use (brand/make).  If the power is not working correctly or is different than what is currently provided, the manufacturer may have implemented the vtx protocol out of spec.

codecae, 2018-02-03:
By adding new levels that are only supported by one piece of hardware, we then break the support for everything else.

Brisingrox, 2018-02-03:
@syrekszymon
No need to change anything for the Stealth Long Range. Works perfect with the script even if the numbers don't correspond with the one in the script. If you go full 700mw and you see 800mw it will still output the correct values on every Mw output.

closed 2018-05-13 by stale[bot]