betaflight/betaflight-tx-lua-scripts #100: Install document update for 1.0

KhrashPadFPV, 2017-12-01:
Given that the older file structure were specific files per radio, it may be worth explaining what the modified installation / script structure is doing in the installs.  A lot of the common explanations have the user pulling a specific zip for their radio and unzipping that.

If (I am still unsure this is the case, this is part hope, part assumption) the installation is just a matter of copying the whole zip, because the script detects your radio and runs the scripts from the right subdirectory, that is worth stating in the documentation, as it is a major departure from previous versions.

Right now the install documentation leaves the user questioning whether it is up to date or whether the the user should be trying to 'figure it out' by pulling scripts from the appropriate subdirectory.   A short sentence or two would alleviate potential confusion, as well as providing a reference point for those who had installed 0.3 (or read about the install such as in Oscar Liang's article on VTX control)

mikeller, 2017-12-01:
@KhrashPadFPV: From 1.0 on, the entire contents of the zip needs to be copied over the SD card. The front page of this repository (https://github.com/betaflight/betaflight-tx-lua-scripts) already explains this and has step by step instructions for installing.

Can you please point out where else you would like to see these instructions?

KhrashPadFPV, 2017-12-01:
Here's the nugget of clarification I think would help (and thank you so much for replying, I have been looking for confirmation all day that this was the case). Here is my stab at an updated paragraph, if that is ok:
"Unzip the file bf-tx-lua-v1.0.zip from the "Releases" link above and drag the contents to your radio. If you do this correctly, the SCRIPTS directory will merge with your existing directories, placing the scripts in their appropriate paths. You will know if you did this correctly if the bf.lua file shows up in your /SCRIPTS/TELEMETRY directory.Notes: - It is no longer necessary to unzip or copy specific scripts for your radio.  After v1.0, all scripts are merged under one folder, and the LUA code selects the right subdirectory automatically.
- The src directory is not required for use and is only available for maintenance of the code. While it may work to use this directory, you may encounter memory issues on your transmitter.
"

mikeller, 2017-12-01:
@KhrashPadFPV: Yeah, that makes it easier to understand, thank you. Do you want to open a pull request for this change?

LightningStalker, 2017-12-27:
There is an `obj` directory at the top level of the zip file. We copy the contents of this directory onto the SD card correct?

mikeller, 2017-12-27:
@LightningStalker: Yes.

closed 2018-05-13 by stale[bot]