betaflight/betaflight-tx-lua-scripts #101: D8 hub telemetry

LuckyFPV, 2017-12-05:
I have a Beebrain V2 that uses D8 hub telemtry, I can see all the telemetry values on my X9 from the Flight controler but the bf.lua does not work. I have the most recent version and does work with other models.
Any help would be appreciated.

mikeller, 2017-12-05:
No surprise there. FrSky hub / D8 does not support telemetry data transfer from the TX to the flight controller, so this can not work.

closed 2017-12-05 by mikeller

LuckyFPV, 2017-12-06:
Thanks  How do they manage the stick commands via the OSD? That works fine. Infact I can see the same  menu options when i invoke the bf.lua that i see in the goggles with the sticks

mikeller, 2017-12-06:
The stick commands are just read by interpreting the stick positions that are sent by RC commands. This has nothing to do with telemetry.