betaflight/betaflight-tx-lua-scripts #104: Everything displays but nothing saves on 1.0.1 - R-XSR

beeb, 2017-12-22:
Hello,

This is not how I like to write issues but I have no real indication on how to reproduce the issue.

I was using v0.3 absolutely fine until next week, on betaflight 3.2.2. The other day, I decided to upgrade the lua to 1.0.1 and betaflight to 3.2.3 on my 4 quads.

All 4 have the same flight controller (hobbywing f4 - target omnibusf4sd). One runs on crossfire and is working fine, one has a FrSky XSR and works fine.

I have problems with the two quads on R-XSR and FusiousFPV Stealth Race. Nothing saves, not even the PIDs and rates, so I'm assuming the issue doesn't stem from the Stealth and must be related to the receiver. These two quads used to work totally fine on 0.3 and they both run the latest firmware 171103 on the r-xsr. Reading the values works fine and I can see information in all tabs of the lua. Only when I hit "save page", the page reloads and displays the old values again. I tried refreshing the page several times but to no avail.

Any idea? Thanks

beeb, 2017-12-22:
Additionally, a friend of mine also has the issue with his modded r-xsr, but no issue with the non modded one. The mod that we both did is basically soldering a wire to the "P" pad and routing this to the TX of an uart : https://oscarliang.com/ctt/uploads/2017/10/frsky_r-xsr_rx_receiver_smartport_sbus_inverted_uninverted_pin.jpg

beeb, 2017-12-30:
In a private message discussion in RCgroups, frsky confirmed the issue.

Clynikal, 2018-01-13:
Was this raised to the BF devs yet?
I’m encountering same issue.

Clynikal, 2018-01-13:
Sorry just saw you posted in BF issues. Never mind.

beeb, 2018-01-13:
Actually no, I didn't post since the guys from FrSky were apparently already in contact. But in the end the suggested to add a 4.7K to 10K resistor between the S.port (I assumed the uninverted "P") and GND pins on the R-XSR to solve it. The other alternative is to wait for FPort to be available and do everything, including telemetry, via a single wire, rendering the smartport mod unnecessary.

mikeller, 2018-01-13:
@beeb: FPort for r-XSR is in the works and should be ready soon, according to FrSky. We'll have to test again when it is available, but it will use the SmartPort pin on the RX, and if there indeed is an electrical problem (with the 'uninverted' pad from what you are describing), then using the pull down resistor might be required with FPort as well. (But then, that's not really a big issue, and can probably be fixed by FrSky in a future hardware revision.)

beeb, 2018-01-15:
@mikeller Thank for your reply. Doesn't the fact that this protocol is uninverted remove the need for the hardware mod on the R-XSR? In such a case, we could use the normal SmartPort pin again, thus removing the problem right?

mikeller, 2018-01-15:
@beeb: Since the inversion on existing RX is done in hardware, it is not possible to change the inversion with a firmware upgrade to FPort. Only yet-to-be-released new RX hardware that is specifically designed for FPort will be uninverted.

beeb, 2018-01-15:
@mikeller OK gotcha! Weird that FrSky suggested FPort as an alternative solution to the pull-down resistor. Couldn't we use an sbus port on the FC with dedicated inverter for this Fport, and use the normal inverted pin on the RX?

Clynikal, 2018-01-15:
[URL=http://s893.photobucket.com/user/HJcrick/media/8F44C3A4-D339-4A27-ABD0-389BA2853956.jpg.html][IMG]http://i893.photobucket.com/albums/ac140/HJcrick/8F44C3A4-D339-4A27-ABD0-389BA2853956.jpg[/IMG][/URL]

mikeller, 2018-01-15:
> Weird that FrSky suggested FPort as an alternative solution to the pull-down resistor. 

I can't comment on that as I don't have the hardware. But if it needs a pull-down resistor for SmartPort, then it will need one for FPort as well in all likelyhood.

> Couldn't we use an sbus port on the FC with dedicated inverter for this Fport, and use the normal inverted pin on the RX?

Yes that is definitely an option. You will need a bidirectional inverter for this though, like so:

[image]

Clynikal, 2018-01-15:
Perhaps (hopefully) frsky will release a firmware update that fixes the telemetry issue as well as enables fport. 
The sPort solder tab has to be used for fPort because it is uninverted and bidirectional, similarly the sbus rx chip on the FC is not bidirectional so a standard uninverted (or a nice inverted) uart will be required. 

Back on topic, I have telemetry writing from my QX7 via LUA after a rx firmware update although it is very unreliable, I continue to get drop outs. I consider this a frsky issue and I will try the pull down resistor next week when I get to a store. This could be fixed in firmware by refining the values for high/low.

Clynikal, 2018-01-16:
I’ve tried the pull down resistor in 4k7, 7k5 & 10k 0.5w resistors. 
The 7k5 was most reliable although it was still unreliable (fail to write after arm). It did reduce the amount of drop outs over a 10 minute period significantly. I will note that all functions will work at some point in time and that is usually after rx and tx restart. 

Sbus remained rock solid. 
FC is DYS F4 Pro. 
Radio is QX7S

I also tried a brand new rxsr flashed with latest fcc firmware and all the same findings were reproduced. 
The greatest chance of failure seems to occur with arming. 

I also tested with a Matek F405 OSD FC and the results were replicated. 

I’m sure BF could do something to work with the reciever but co soldering these issues do not occur with other frsky x rx’s I think it is safe to conclude this is a rxsr issue. 

I’ll post a video later.

mikeller, 2018-01-16:
The fact that arming changes something here is weird - arming should not change anything to do with the serial connections between RX and flight controller, and the RX does not even know if the flight controller is armed or not. Do you use stick arming or switch arming?

Clynikal, 2018-01-16:
I think it’s a noise issue more than anything, I should probably try a cap to clean it also. 

I use a sticky logic switch between H & F

beeb, 2018-01-16:
@mikeller thanks for your previous answer about the hardware inverter. I'm not sure if most F4 boards have bidirectional inverters on the SBUS pin, but I kinda doubt it. Do you know if this is common practice or if all F4 boards with hardware inverters on the SBUS pin are unidirectional?

beeb, 2018-01-16:
@Clynikal @mikeller what I'm still wondering is why this issue does NOT happen on 3.1.7. Both myself and FrSky confirmed we had working LUA with the modded R-XSR (no pull-down resistor) and BF 3.1.7. I downgraded one of my problematic quads and sure enough it did work.

mikeller, 2018-01-16:
@beeb: Often the inverters for SBus are unidirectional. But some boards have a bidirectional inverter on the port that is designated for SmartPort.

The timing of SmartPort packets was changed between 3.1.7 and 3.2 as a side effect of some unrelated changes.
3.3 includes a fix (betaflight/betaflight#4863) that ensures that the timing is acceptable for all RX. Maybe you should try 3.3?

Clynikal, 2018-01-16:
@mikeller nice merge. I really hope you’re right, I’m tempted to try 3.3 just for this, likely on my thrasher tomorrow. 
I’ll report back.

beeb, 2018-01-16:
Very nice, I'll try 3.3 if I get time next weekend, the only fear is that my quads with R-XSR are not exactly trashers... ^.^"

Clynikal, 2018-01-17:
@mikeller I updated to 3.3 using build 522 on a matek F405 OSD and encountered the exact same issues; read sometimes, write sometimes, unstable all the time. 
I connected via both the univerted pad and inverted to soft serial. 

It’s worth mentioning I purchased 3 of these rx’s at once so there’s a small chance it’s a batch. 

From here I’ll grab a XSR or X4R and test but that won’t be for a while. 
Do you recommend reflashing opentx and or LUA? Sorry I don’t mean to use you as troubleshoot.

Clynikal, 2018-01-17:
I’ve just reflashed the rx with latest stable firmware and when i just had 5v and GND connected only I was still get telemetry lost alerts. This can’t be BF.

beeb, 2018-01-17:
@Clynikal If you're standing too close to the RX with your TX this can happen. I wouldn't be worried about that too much. Did you try reverting to BF 3.1.7 see if like me, it solves the problem?

Clynikal, 2018-01-17:
@beeb I think you’re dead on. How did I miss that???
@mikeller sorry to waste your time mate I’ve tested at a suitable distance on both 3.1.7 & 3.3 and both are good.

mikeller, 2018-01-17:
@Clynikal: No problem. :-)

tavdog, 2018-02-16:
I'm seeing the same issue.  Trying to save only certain values fails.  Eg. saving VTX to channel 2 or 3 works fine, but will not save when setting VTX to channel 1.

beeb, 2018-02-16:
@tavdog are you talking about 3.3 RC1 ?

tavdog, 2018-02-16:
No I haven't upgraded to 3.3 yet. I will try and update results here.

Clynikal, 2018-02-16:
I’ve had success with all 3.1.x, 3.2.x & 3.3 with R-XSR in all telemetry screen now. 
Granted it doesn’t always work but it’s I’m not laying that at BF’s door. 

Make sure you upgrade to latest frsky firmware at the minimum. I’m expecting you’ve tried with different VTX’s and FC’s as well as using the OSD.

beeb, 2018-02-16:
@Clynikal the problem only occurs if the infamous "inverter mod" was performed on the R-XSR. I don't know if that was the case for you when using it successfully with BF 3.2.x.

@tavdog this should fix it for you I think

Clynikal, 2018-02-16:
I ran the R-XSR using the inverted tab today on 3.2.something, I only encountered intermittent issues about 10% of the time. They feel like link errors when it occurs but it’s non a constant or consistent issue.

tavdog, 2018-02-17:
I updated to 3.3.0 today.  I am experiencing the same problem.  R-XSR (normal setup, no inversion hack), Matek F405 with FCHUB-VTX, BF LUA Taranis.  Its just so odd that it only fails when trying to update to certain numbers.  Eg.  Updating to Band R, Channel 2 works all the time.  But updating to Band R, Channel 1 never works.  Updating to Band F, Channel 1 works all the time, etc.  So weird.  Actually Band R, Channel 1 works sometimes on the very first attempt after full reboot.

beeb, 2018-02-18:
This rings a bell, I remember also having problems with Raceband 1 a while ago, but I'm not sure which version of BF I was on.

ttjs82, 2018-02-26:
I have the R-XSR receiver installed with DYS F4 Pro using the F.Port. I've connected it using the uninverted solder pad on Rx to the TX1 on the flight controller. Needs have the serialrx_halfduplex = ON in CLI to work. Radio control, telemetry and Lua scripts seem to be working just fine, also saving the Lua script settings.

beeb, 2018-02-26:
@ttjs82 thanks for the feedback! Can you make sure that you can switch from Raceband whatever to Raceband 1 with the LUA script, provided you have smartaudio or tramp telemetry setup?

ttjs82, 2018-02-26:
Unfortunately I don't have any smartaudio or tramp capable vtx available for testing currently. So I was only able to test saving pids and such to be exact.

beeb, 2018-02-27:
@ttjs82 Ah ok, that seems to be mostly solved except for what I mentioned above. Still, it's way better now with BF 3.3 than it was on 3.2

DaEmpty, 2018-04-17:
I can't save channel R2/R3 with BF3.2 and BF3.3 with my R-XSR. So, it should not be related to BF3.3 only.

beeb, 2018-04-17:
Hello @DaEmpty , yes the problem was most prevalent on 3.2. It is better on 3.3 where only a few channels are not working. But 3.2 was clearly broken.

IB1387, 2018-04-23:
I don't know if this is old news for y'all, but I've also got the R2/3 not being saved error and tested my lua script issues today, so here are my findings:
I use the LUA script just for vtx control, so that's the only thing I've tested. I noticed something is wrong when trying to save RB2/RB3 to my TBS Unify HV Race via the LUA script on my Taranis X9D+ and my Omnibus F4 using a R-XSR. I had to use the OSD to switch channels on a big race instead of LUA, which was a big hassle for me, but that worked perfectly, so I think we can rule out the smartaudio implementation. 
I tried different versions of the LUA script (v0.3 and the most up to date one): Both can't save RB2/3. I've tried the most up to date versions of LBT and FCC firmwares for the XJT and R-XSR with no effect. None of the firmwares are able to save RB2 or 3. 
The only thing I've found that helps is flashing the R-XSR with the FPort firmware (at least with the EU one it works), and reconfiguring the FC to use FPort. Now I can succesfully save all channels on my Unify. 

Hope these findings can help. Seems like a bug in the R-XSR firmware implementation? Sorry if this was already known. Just tested today to get it working for next weekend and thought to share my findings. Let me know if I can do anything for y'all.
Cheers!

kreidler, 2018-04-25:
@IB1387 thank for the FPORT idea. Struggling the same issue and tried FPORT. Stick controls are possible but I am ending up in a non bidirectional serial like described in https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol no.7. So, switched back to OSD controls :(

BTW: Nils, great 2nd place :smiley:

kreidler, 2018-04-26:
@IB1387 : Thanks, I use the HGLRC F440 stack with R-XSR.
Neither inverted nor uninverted Smartport works fully with UART1 (normally SBUS pad) -> Normal Smartport connected to UART1 allows stick movements and reciever based telemetry only. Uninverted hack completely without function. All CLI options tried.
Using pad on Rx together with TX6 as 'F4 with 'uninverted SmartPort' hacked receiver' full telemetry and LUA functionality available. As TX6 was used for ESC sensor with unaccurate voltage reading only FPORT will be the new choice.

OT: Self attending as mechanic for my son.

closed 2018-06-02 by stale[bot]

alfskaar, 2018-10-21:
XSR-M is working with R2, R3 can we have this reopen and see if it can be fixed for the R-XSR using SBUS and SmarPort