betaflight/betaflight-tx-lua-scripts #106: Script syntax error BF/ui.lua:335

benwap, 2017-12-28:
Hi everybody,

After I updated  my X9D+ Taranis in opentx 2.2.1 I have been getting a memory error. It seems to happen after paging the fivth lua screen VTX. The screen writes the following warning:

Script Syntax Error
BF/ui.lua:335
not enough memory

I tried many new updates and couldn't arrived to solve this issue.
I need your help.
Thanks!
Ben

jrwrodgers, 2017-12-30:
Hi @benwap, I'm having the same issue too. "not enough memory" error message as yours above. 
Still no solution yet but will update here if I find anything.
Thanks
James

raphaelcoeffic, 2017-12-30:
See #97

ethomas997, 2017-12-30:
Looks like the available memory for 'lua' scripts was reduced in OpenTX 2.2.1.  Hopefully they'll do an update before too long; looks like a fix is in the works:  "PR: Save memory (move usb mass storage buffer to reusableBuffer)" https://github.com/opentx/opentx/pull/5579

I was thinking we could also see about reducing the memory footprint of the 'VTX' scripts by changing the 'freqLookup' table to a function with case/switch-type statements.

closed 2017-12-30 by mikeller

benwap, 2017-12-30:
Ok! thank you for the response. It illuminates a bit my mind on the issue. So we just have to wait for a new update from open TX...
Best reguards and wishes

limonspb, 2018-02-23:
So maybe this will help for somebody.
I have same issue with OpenTX 2.2.1, but for model number 4.
**Script Syntax Error
BF/ui.lua:335
not enough memory**
once I setup everything for model N1 - this "bug" disappears.

oranmike28, 2018-04-01:
@raphaelcoeffic 
I did go to #97 and use the V1.0.1. It works but only on the model active when turning on taranis, changing model and then the script doesn't work on all the models. Various errors occur; not enough memory(335 or 10), telemetry protocols not support(40).

Lazy8onez, 2018-08-07:
I also tested it with opentx-x9d+-noheli-lua-en-2.2.2 and BF scripts: bf-tx-lua-1.0.2.zip and havr this issue.
Thing is, if I go directly to the lua script page after powering the tx ON, it works great. But, if go to that lua page after I, for example, edited settings on my model, it shows this error on page 1.