betaflight/betaflight-tx-lua-scripts #107: Feature Request: 'back button' functionality / button navigation changes

csurf, 2018-01-06:
Can we reorganize the button navigation & add some kind of 'back button' functionality for page scrolling? 

I'd vote for the following changes overall...
```
'MENU' button: 
- short-press: page forward
- long press: page back

'ENT' button: 
- short-press: normal operation
- long-press: open the 'Menu: save page' menu 

'+' (or '-') button:
-short-press: normal operation
- long-press: open the 'Reset Telemetry' menu 
```
It's just my opinion, but it seems like this setup would make things a bit more intuitive and quicker... 
(Please don't flame if you disagree...I just figured I'd try & suggest it)

Ideally, long-pressing the menu should scroll back to the previous page, but that is currently being used to open the 'save page' menu. 

sound reasonable?

codecae, 2018-01-07:
Long press of ENT is reserved in Open TX.  We can't override it.  Furthermore, there aren't any long press events for + or -

csurf, 2018-01-08:
understood... closed.

closed 2018-01-08 by csurf