betaflight/betaflight-tx-lua-scripts #112: Tramp HV, unable to set power to 200mw

paulo-serrao, 2018-01-28:
I'm having a very odd issue while trying to configure my Tramp HV using these LUA Scripts. It seems that some combinations of settings always fail to save. I set the options I want, longpress menu to save, it blinks "Saving" a couple of times, then displays "Retrying" and after a while it gets back to the VTX setup screen. Reloading the settings shows that the VTX wasn't updated to the new settings...

Here are some real examples:

power band channel 
100mw R 1 OK
100mw R 2 Not OK
100mw R 3 Not OK
100mw R 4 OK
100mw R 5 OK
100mw R 6 OK
100mw R 7 OK
100mw R 8 OK

200mw R 1 Not OK
200mw R 2 OK
200mw R 3 OK
200mw R 4 Not OK
200mw R 5 OK
200mw R 6 OK
200mw R 7 OK
200mw R 8 OK

Does this happen to anyone else? Is there any way to debug this?
I am forced to go to the OSD and chose those settings I can't via Lua, and they work fine if set that way...

RX is a R-XSR with latest International FW (R-XSRFCC171009), FC is DYS F4 Pro running BF 3.2.3

BigRad01, 2018-02-05:
I'm also seeing this issue using Lumenier TX5GS (Tramp Protocol) and Frsky R-XSR with FCC 171009 firmware.  OpenTx 2.2.1 with SD 2.2V0013.  Lua 1.01.

I can set some frequencies, but others fail as described above.  Haven't checked OSD.  I've been using 25mw power:
25mw R 1 OK
25mw R 3 Not OK
25mw R 8 OK

Footr, 2018-02-07:
Also having this issue on my Tramps.  Its also showing up on the Amoway VTX04.  You can change frequencies via the OSD and that works just fine.

paulo-serrao, 2018-02-08:
Are there any instruction on a way to debug this?

codecae, 2018-02-08:
Best way is with a logic analyzer.  The osd and the lua feed the same parameter group on the FC, so they should behave identically.

Wish I had a tramp to test with :/

paulo-serrao, 2018-02-08:
I don't own a logic analyzer, so I guess i'm out of luck :( 

Apparently its an issue with the Tramp protocol and not the transmitter (Tramp HV) itself.

Isn't it possible to somehow send the commands and have them written as part of the FC blackbox?

codecae, 2018-02-08:
I wouldn't say you're out of luck.  Just need a little time to get to the bottom of it.

codecae, 2018-02-08:
If you're using a r-xsr, make sure its firmware is up to date.  Also, try using a more recent build of Betaflight 3.3 from ci.betaflight.tech.

paulo-serrao, 2018-02-08:
Using R-XSR with latest International FW (R-XSRFCC171009) as described on first post.

Haven't tried with BF 3.3 yet. Since choosing those channels work from the FC OSD, I'm pretty sure its an issue somewhere between the Lua and FC.

codecae, 2018-02-08:
Its likely an issue with the link.  Please try 3.3, as there are updates to timing in smartport that were built around issues with the r-xsr in mind.

spacebubi, 2018-02-24:
Running on CRSF and 9XDplus all updated, on my 2x F3 Fc (a Radiance Dshot and a Lux V2) both not working Vtx Page. CRSF is on UART TX1 (works all telemetry) and T cable is on TX3 and on script Vtx page appear that:

Band       U          DEV    Tramp
Channel    0        freq    0
Power    0
PIT        OFF

Any solutions? i've tryed by SOFTSERIAL but unsucces...LUA after that dont see nothing on DEV

spacebubi, 2018-02-24:
SOLVED in my case using UART 5 on LUX V2 , reading manual its tagged "for telemetry". 
IRC Vtx  works all now like other pages.

omenus, 2018-03-01:
I can confirm this same problem with BF 3.2.5 and r-xsr cant save RB 2, 3 and sometimes other channels over LUA 1.0.1 - so this problem exists.

paulo-serrao, 2018-03-02:
Same issues with BF 3.3 just released @codecae 
Using latest R-XSR fw as mentioned in the first comment

KenMurphy, 2018-03-30:
My setup is R-XSR with the FPort firmware running betaflight 3.3 on a RDQ bardwell board. OpenTX 2.2.1 on X9D+, latest lua scripts. I deleted all the crossfire lua scripts from my radio to prevent OOM errors.

I just did a bunch of channel/power change testing including the ones mentioned in the original post. It switched instantly every time and did not throw any errors. Confirmed that it actually switched by checking video through my goggles every time.

codecae, 2018-03-30:
Try [this version](https://drive.google.com/uc?authuser=0&id=1K5bsbJPHgzx6awizgWDLcDjFRSdraSJ1&export=download)

codecae, 2018-03-30:
Unrelated note for @KenMurphy -- Deleting the crossfire scripts will do nothing to help your memory situation, as they only occupy memory when they are running.  If they are sitting on your SD card, they are not actually occupying any memory on the transmitter.

KenMurphy, 2018-03-30:
@codecae That is what I thought but after deleting them I haven't gotten the OOM error once. Before I consistently got the OOM error every single time after reaching the 3rd page. Deleting the CF files is the only thing I changed.

paulo-serrao, 2018-03-31:
Hi!

I copied the content of obj/SCRIPTS into my Taranis SCRIPTS folder, but I
still have the same issue. Can not save Raceband 200mw C1 and C4.

Are you sure you sent me the right file? The only changes I see are on the
ui.lua

compudaze, 2018-04-18:
I'm having the same issue with Matek VTX and Lumenier VTX using the tramp telemetry protocol. I've updated the VTX firmware to latest v1.9.

Pyrodrone F4 OSD FC (has hardware inverted for Telemetry s.port)
Betaflight 3.3.1
R-XSR R-XSRFCC171009.frk

closed 2018-05-25 by stale[bot]

paulo-serrao, 2018-08-17:
Did anyone ever found a way to fix this?

KenMurphy, 2018-08-17:
@paulo-serrao Run fport instead of SBUS