betaflight/betaflight-tx-lua-scripts #113: Raceband 2 +3 not saved

derdanu, 2018-02-07:
Setup:
Furious FPV Fortini OSD (BF 3.2.4
Furious FPV Stealth (improved Version / new connector) 
Frsky X9D + opentx 2.2.0
Lua 1.0.1 

Changing VTX channels works fine, but only Raceband 2 and 3 were not saved. The Script tries to retry the process but fails again. 

Any way to provide a debug log? 

Transmitter and LUA combination works well on other Setups.

derdanu, 2018-02-07:
Similar like #112 as it uses the protocol, but I run R-XSR as well. BTW tested all firmwares provided by Frsky.  The other working Setups don't have an R-XSR receiver.

codecae, 2018-02-07:
If you're having issues with the r-xsr, please try using a more recent build of 3.3 from ci.betaflight.tech

derdanu, 2018-02-08:
Same issue with 3.3 Build #583 https://ci.betaflight.tech/job/Betaflight/583/

codecae, 2018-02-08:
I unfortunately do not have a tramp-based VTX so I have no way of reproducing this issue.  Hopefully someone else has one and can chime in.

derdanu, 2018-02-12:
@codecae please drop a line at Tech@furiousfpv.com and ask for Tommy Schintgen.  Please provide my name in the conversation as I talked with him. Maybe he can send you a Stealth VTX for testing and further development.

Sybalau, 2018-02-16:
I have the same isue with Stealth Race VTX. Raceband 2 and 3 don't saved.

codecae, 2018-02-16:
@derdanu Thank you for the follow-up!  I will reach out now.

codecae, 2018-02-16:
@derdanu I have written an email to the suggested email and am awaiting a response.  Thanks for your help!

eumelOne, 2018-02-17:
I also have observed this issue with a unify pro hv race edition. Only RB2 and RB3 won't be saved...

derdanu, 2018-02-17:
@eumelOne I can confirm this. A mate has the same issue with an unify nano.
It defintily a R-XSR bug. I have switched on of these Setups to a crossfire nano RX. Saving Channel 3 works like a charm.

Brisingrox, 2018-02-17:
I have tested this issue with FortiniF4/FortiniF4osd rev1+rev2 but with the SPD15 Receiver and didn't encounter any issues.
Tested with Steahlt Race and Long Range without issues

gregory-gauthier, 2018-02-28:
same issue, can't save Raceband 2 & 3

gregory-gauthier, 2018-02-28:
Same issue, I can't save Raceband 2 & 3
VTX : Unify Pro HV Race
RX : R-XSR (firmware 171103)
FC : LUX V2 & Micro Lux V2 (Betaflight 3.2.5)

ps915, 2018-03-10:
Same for me.

KISS FC V2 
ImmersionRC Tramp

I could change channels to RB3. I replaced it with a TBS Unify and i worked!

dbertschi, 2018-03-25:
Same issue for me.
Fortini F4, RSXR 171103, Furious Stealth VTX. BF 3.3.0.
On the OSD i can change to all RaceBand channels - but vith the latest LUA Script on my X7, RB2 and RB3 can't be saved.

Barux, 2018-03-25:
Same here running BF 3.2.0, OpenTX 2.2.1, latest lua V1.01, R-xsr, Stealth race.
Can save with BT app and via Pidflight vtx

Any solutions?

David-VG, 2018-03-25:
Just while solved, try FPORT.

https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol

DaEmpty, 2018-04-18:
Same for me
Kiss FC V1 BF3.3.1
R-XSR (tried all versions)
Unify Race

With the normal XSR everythings works fine

Barux, 2018-04-18:
> It works with F-Port protocol

Yes guys, FPORS solved all lua script problems I had

closed 2018-05-25 by stale[bot]

rustbg, 2018-07-18:
Same here. R-XSR, BF 3.4 stable. R2 and R3 are not saved. 
Using latest lua script with the dsetpoint fixed. 
Tried so far:
R-XSR on LBT and FCC
F-Port (un-inverted)
Un-inverted s.port + inverted sbus
Inverted s.port on softserial + inverted sbus

Cookiedealer, 2018-07-18:
Thank you very mich for all Information!
Hello fro.e North Germany
FlipFlopRacer

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/betaflight/betaflight-tx-lua-scripts/issues/113#issuecomment-405824028",
"url": "https://github.com/betaflight/betaflight-tx-lua-scripts/issues/113#issuecomment-405824028",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [betaflight/betaflight-tx-lua-scripts] Raceband 2 +3 not saved (#113)",
"sections": [
{
"text": "",
"activityTitle": "**rustbg**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@rustbg",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"betaflight/betaflight-tx-lua-scripts\",\n\"issueId\": 113,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/betaflight/betaflight-tx-lua-scripts/issues/113#issuecomment-405824028"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 301328895\n}"
}
],
"themeColor": "26292E"
}
]

David-VG, 2018-07-18:
@rustbg I dont have this issue with FPORT, are you sure you have?

rustbg, 2018-07-19:
Fport was even worse for me, as PID page was completely not functional :-(
Best result is uninverted (hacked) s.port, where just r2/3 don't work, but everything else is ok, even some say retrying, but eventually do save after a bit more time.

Benaresq, 2018-11-22:
I just semi-resolved this issue by changing to Fport. The channels save now, but the script always says "Retrying" after each channel change.

DaEmpty, 2018-11-22:
FPort requires a bidirectional inverter on your fc. if you have e.g. an omnibus f4, you are lost. Smartport does not support vtx channel 2+3, and FPort is not working when you want to use the plug of the r-xsr. And even if you solder a cable to the uninverted signal pad, you get retrying messages all the time.

Ignoring such a major bug for almost a year now, is a clear statement from the betaflight development team. The OSD works fine, it's just not very convenient for changing vtx channels. But at least it does it's job.

mikeller, 2018-11-22:
@DaEmpty: Why don't you submit a fix for the bug that you think there is, instead of just making accusations of bugs being 'ignored'?

Benaresq, 2018-11-23:
@DaEmpty I've got telemetry working and the LUA scripts all appear to be fine, I'm running a custom script to change channel from a switch that seems to work OK.

When I was running Sport, if I had the Channel LUA page open, it would update when I hit my channel change switch, but now I have to select update to see that it's changed. On the plus side, Raceband 2 and 3 work now, so it solved the problem I was having.