betaflight/betaflight-tx-lua-scripts #117: Fport not working ?

dark-nico, 2018-03-05:
Hello, 

I'm trying to make the LUA scripts (1.0.1) works with Fport on a R-XSR ( with betaflight 3.3).
I'm getting the following error message "Telemetry protocol not supported!"
On the wiki (https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol) it says it should be working. 

Am I missing something ? What can I do to help ? 
Ps : I'm on the last version of OpenTX (2.2.1) on a X9D+

dark-nico, 2018-03-05:
Ok I enabled halfduplex and redisabled it, and now it's working both read and write. weird.. 
But the rate page 1/2 stays empty, not error but no value. 
I suspect it's related to the recent change in betaflight 3.3. 

> The way that rates are configured has been changed (#4973). 
Betaflight now supports independent rates settings for roll / pitch / yaw. 
When updating from an older version of the firmware, make sure to convert your settings to the new parameters as follows: 
rc_rate becomes roll_rc_rate and pitch_rc_rate, 
rc_rate_yaw becomes yaw_rc_rate, 
rc_expo becomes roll_expo and pitch_expo, 
and rc_expo_yaw becomes yaw_expo. 

>

FPWibi, 2018-03-09:
Hello,
I have exactly the same problem. Frsky XSR Receiver with FPort Firmware und Betaflight 3.3. 
At first all pages in the script worked. Then I changed the rates in the script. The changes were made but there are no more values on the rate page 1/2. Every other page is working fine.

JaYp247, 2018-03-27:
I must be doing something wrong. I can't even get telemetry data to populate. Would you mind sharing some screen shots of your Port and Configuration pages or link a setup guide.

      The way that rates are configured has been changed (#4973).
      Betaflight now supports independent rates settings for roll / pitch / yaw.
      When updating from an older version of the firmware, make sure to convert your settings to the new   parameters as follows:
      rc_rate becomes roll_rc_rate and pitch_rc_rate,
      rc_rate_yaw becomes yaw_rc_rate,
      rc_expo becomes roll_expo and pitch_expo,
      and rc_expo_yaw becomes yaw_expo.

closed 2018-05-13 by stale[bot]