betaflight/betaflight-tx-lua-scripts #120: VTX Settings over Taranis X9D - only works once then not displayed on Taranis anymore

RipperDrone, 2018-03-25:
verified on 2 different FCs / setups: Once I delete all settings in the diff file and reset the FC, I get the VTX settings displayed properly on my Taranis (using 1.0.1 Lua Script) and can change them.

After saving, script says saving... then Retrying... then exits. Seems the settings are stored, though (I see in next diff dump what I entered). 

However, after trying to change the VTX settings again, they are not even displayed on Taranis anymore.
Verified with both Furious and Akk VTX - same problem...

So 2 issues, it seems:

1. Settings saving once only, then becoming invisible for Lua Script
2. Saving VTX settings always throws a 'retrying' message

codecae, 2018-03-25:
Which version of Betaflight?

RipperDrone, 2018-03-25:
BF3.3 official

kptmorhen, 2018-03-31:
I have almost the same issue. BF 3.3. Opentx 2.2.1, crossfire nano, taranis x9d+, tbs unifi. When i switch on everything, i can see all telemetry info in bf lua including vtx info. When i try to change vtx params and save it, it works but then there is only "--" instead of values in vtx tab. Changing PIDs work well.

RBeaubien, 2018-04-22:
I'm having a very similar issue.  BF 3.3.1, OpenTx 2.2.0.  I turn radio on, then plug in battery on quad.  Telemetry comes up and I am able to edit PIDS, VTX, etc.  Unplug quad, to put new battery on, Telemetry comes up, but all script data screen do not show any data.  Telemetry page shows all sensors updating as well as battery voltage updating on main screen.  RX is an R-XSR.

codecae, 2018-04-22:
R-XSR RXs are notorious for performing poorly.  There's nothing about the scripts that treat them ny different than any other RX.  Make sure your firmware is up to date or maybe try a different RX.

RipperDrone, 2018-04-22:
I think it's related to over-populating the telemetry frame with too many sensors. Try removing all but 4-5 sensors, and it will maintain flawless transmission. It seems the R-XSR is more sensitive to timings than other RX so sensor messages seem to be bogging the transmission pipeline after some time has passed since last reset...

RBeaubien, 2018-04-22:
That seems to help.  Still fails sometimes, but much better.  Thank you.  Is the control from the scripts related to any of the telemetry sensors (F000 perhaps)?

codecae, 2018-04-22:
@RBeaubien The scripts communicate without any issue with other FrSky RXs.  The R-XSR is really the only offender.  

The way the scripts work is no different for each RX.  The data is sent identically across the s.port stream with an R-XSR as it is for a X4R-SB, for example.  If there were issues with the scripts, specifically, the same issue would be present on the X4R-SB, as well.

The issue appears to be frame loss or corruption.  The scripts are sending out requests for information but they are getting lost somewhere between the quad and the transmitter.  No matter what hardware you're using, the frames are build identically and are dispatched using the LUA telemetry API in OpenTX.  The way this works has not changed in the newer versions of the scripts. The API simply sends it out into the air and expects the RX to receive and hand it off to the FC.

The FC handles MSP requests identically for all S.Port frames and responds identically as well.  There's nothing R-XSR-specific here either.

There are a variety of reasons why this could fail to work... Buffer sizes too small, timing issues, refresh intervals... lots of things.

codecae, 2018-04-22:
@RBeaubien Telemetry sensors are receive only (by the transmitter).  There aren't any sensors that are used for the scripts.

codecae, 2018-04-22:
@RipperDrone Interesting that removing sensors has an effect, as they shouldn't affect the outgoing telemetry buffer, nor should it prevent the FC and RX from transmitting them.  I've updated an R-XSR to the latest firmware to see if I can recreate the issue.  Can you provide me with specific sensors that you are removing so I can attempt to recreate?

David-VG, 2018-04-22:
I think i dont have this issue with FPORT. Do you want me to try something more specific?

codecae, 2018-04-22:
FPORT doesn't use SPort -- this is probably why it's okay.  The issue only seems to be present when using SPort telemetry.

David-VG, 2018-04-22:
I know. Thats why i suggested to try.

RipperDrone, 2018-04-22:
@codecae some time back, I have at least improved the R-XST tele issue a lot by removing a lot of sensors less interesting to me like accx, accy, accz, F0000 and fuel (as far as I remember). Tried to keep VFAS, A4 (single cell voltage), RSSI, curr, hdg. On some builds, I have GPS as well - this made it even harder for the R-XSR to keep up with the frames / timing since it was even more sensor data. 

When monitoring the asterisks blinking on the Taranis X9D telemetry screen whenever a sensor is being updated, I could clearly see that after each reboot, all sensors would get frequent refreshs whereas after a while, some of them slowly tapered off, up to a point of sensor lost messages...

RipperDrone, 2018-04-22:
I have done some experiments with deleting different subsets of sensors - couldn't identify a clear priority or sequence. It just seemed to be related to the ampunt of sensor data and maybe - as you mentioned - some buffer overflow over time...

mikeller, 2018-04-22:
@codecae: Since the MSP tunnelling uses sensor Ids and sensor data slots, and assuming that the RX is trying to keep track of what sensors are responding, I can see how an overrun of the list holding the active sensor Ids would cause problems with MSP tunnelling. One main difference between the r-XSR and other FrSky RX is that it uses an 8051 MCU (integrated into the CC2510 RFM). This has considerably less resources than the STM32 MCU used on other RX, so it stands to reason that supporting the same number of sensors that the other RX support is probably not going to work properly.

mikeller, 2018-04-22:
Re FPort vs SBus / SmartPort: Since FPort requires the RX to only support a protocol handler / queues for one UART instead of 2, I can imagine that this probably improves the resource shortage of the 8051 a bit.

codecae, 2018-04-22:
@mikeller  is the R-XSR aware of which sensors are being reported to the user on the transmitter?  Just trying to understand how disabling them would have any effect on the overall bandwidth.

mikeller, 2018-04-22:
@codecae: As far as I know there is currently no provision for the TX to push the list of sensors that have reporting enabled to the RX, I suspect the RX is just keeping a list of all of the sensors that are ever reported on the SmartPort 'bus'.

RBeaubien, 2018-04-23:
I added a feature request to Betaflight to have the ability to get rid of some of the telemetry items: https://github.com/betaflight/betaflight/issues/5748

closed 2018-05-30 by stale[bot]