betaflight/betaflight-tx-lua-scripts #121: Script NOT working

GanceNag, 2018-03-25:
Hello!

I have a problem with this lua script (1.0.1) and my Taranis X9D+ Special Edition (OpenTX 2.2.1).
I had the famous RAM problem so I installed OpenTX 2.2.1 with noheli, lua and luac option in order to solve this.
But now I can't even start the lua script anymore.
I tried reinstalling the SD content, I tried to rediscover the FC sensors but nothing works.
Here's the error:
`Script syntax error
?:0 attempt to index field 'protocol'`

What can I do?

GanceNag, 2018-03-25:
I downloaded 'obj' folder and copied/merged "BF"  and "SCRIPT" folders in the Taranis SD card

AJZOMBIE, 2018-04-02:
script systax error
bf/ui.lua:335
not enoughmemerry

slawek89, 2018-04-14:
I probably know what is wrong. In firmware options should be active both lua and luac.

ethomas997, 2018-04-14:
@codecae You may want to add a note about those OpenTX lua/luac options to the 'Readme.md'

closed 2018-05-21 by stale[bot]