betaflight/betaflight-tx-lua-scripts #122: Script not populating in d8 mode.

turboed13b, 2018-03-28:
Everything works fine in d16 mode but there is a problem with my rx so I have to use d8 mode. Telemetry data shows up but script does not populate.

codecae, 2018-03-28:
There's nothing specific about the script that could address this issue.  OpenTX supplies a generic function for pushing telemetry data to the FC.  If this does not work in D8 mode, there's nothing that can be done in the script to make it work, unfortunately.   I suggest reaching out to OpenTX regarding this issue.

mikeller, 2018-03-28:
D8 (FrSky hub telemetry outside of OpenTX) is unidirectional and does not support sending of telemetry releated data from the TX to the flight controller, so this is a limitiation of the D8 protocol, and cannot be fixed.

closed 2018-03-28 by mikeller