betaflight/betaflight-tx-lua-scripts #126: FPort Compatibility

SPlissken, 2018-04-23:
Is it possible to get this to work with FPort or are there limitations compared with Smartport?

David-VG, 2018-04-23:
It works out of the box.

SPlissken, 2018-04-23:
Thanks, I must be doing something wrong then as it's not working for me at all but worked perfectly under Smartport for several months.

No hardware has changed and control link is fine and I'm receiving telemetry normally under FPort.

I normally use this excellent script with INAV but after moving to FPort it stopped working so I did a clean install of BF 3.3.1 on a SPF3 for testing purposes but it still didn't receive any PIDs.

I'm using:
R-XSR with FPort LBT F/W 180228
SPF3 clone with UART 3 TX connected to R-XSR
X7 with opentx-x7 v2.20

serialrx_provider = FPORT
serialrx_inverted = ON
serialrx_halfduplex = ON

# dump
# Betaflight / SPRACINGF3 (SRF3) 3.3.1 Apr  3 2018 / 10:04:18 (4b7eaf79d) MSP API: 1.37
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource SONAR_TRIGGER 1 B00
resource SONAR_ECHO 1 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
[... 520 lines cut ...]
set d_yaw = 20
set p_level = 50
set i_level = 50
set d_level = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

closed 2018-04-23 by SPlissken