betaflight/betaflight-tx-lua-scripts #134: Frsky X-lite not supported

jafc76, 2018-05-25:
I just received the Frsky X-lite and after to install the LUA Scripts for Betaflight, I'm getting this error:

Script syntax error
BF/radios.lua:43
Radio not supported: xlit

I know that it's a new radio so waiting for the compatibility with it.

Thanks in advance.

raphaelcoeffic, 2018-05-25:
@jafc76 there are certain unknowns as-of the available navigation keys inside the script (being sorted out in OTX dev chat right now). Once this is done, I might make the necessary adjustments.

raphaelcoeffic, 2018-05-25:
@jafc76 could you please test the code changes I've just made? Please bare with me, I have no way to test anything...

jafc76, 2018-05-26:
raphaelcoeffic, Thank you very much!!! It works!!!

The small joystic is working like this. If I move the joystick to the right, the cursor is moving down to the next option inside of that screen, if I move to the left, the same but going up inside of the same screen.
If I move the joystick up, I go to the next page, from PID screen, to rates1, rates2, etc... but if I move the joystick down to go to the previous page, doesn't go back.

Anyway, at the moment, it works, maybe a little tricky how the joystick works but it's functional. Thank you very much again!!

raphaelcoeffic, 2018-05-26:
Can you please confirm if you can reach the menu (long-UP)?
“Previous page” with the down key can be done easily.

jafc76, 2018-05-26:
To go to the LUA screen I have to press a long-down.

Inside of the LUA screen if I make a long-up I get one screen with this message:

Menu: save page
             reload
             reboot

raphaelcoeffic, 2018-05-26:
@jafc76 glad to hear it works properly!

raphaelcoeffic, 2018-05-26:
@jafc76 I made a small modification to support "previous page". The keys have been slightly changed to fit that logic:
- next page: DOWN
- previous page: UP
- pop-up menu: long DOWN

Would you mind re-testing?

jafc76, 2018-05-26:
I just copied the files to test it and works like a charm!

long-down to go to the LUA script.

right to go down and left to go up inside of each screen.

down to go to the next page and up to the previous one.

long down pop-up menu

Don't worry about to test it, I really appreciate your time and work!!

Eveything is good now, but if you change to move down pressing down, up pressing up, next page pressing right, previous pressing left, will be perfect!! Only if you can, no pressure.

Thanks again!!!

raphaelcoeffic, 2018-05-26:
Ah ok! So basically switching both axis? No problem ;-)

jafc76, 2018-05-26:
Yes! I think will be more intuitive to navigate through the menu. You are the man! :)

jafc76, 2018-05-27:
Working!! It's perfect in my opinión.

The only think I don't like is the pop-up menu. Will be better cleaning the background but anyway at the moment is just fine!!

[image]

Thank you so much!

raphaelcoeffic, 2018-05-27:
I believe that one is an issue with OpenTX itself, as it works on X7 correctly. I’ll deal with it ;-)

raphaelcoeffic, 2018-05-27:
https://github.com/opentx/opentx/issues/5912

jafc76, 2018-05-28:
Working! Thank you very much!
[image]

closed 2018-05-28 by raphaelcoeffic

ps915, 2018-06-23:
I tested the latest LUA Script with OpenTX **2.2.2 RC2**.

Navigation is just fine but the **"save page"** dialog is on **"long right"** button.

Wouldn't it be better (more intuitive) to use the **"Function/Shift Button"** to get to the menu?

Here is a video, how we set the buttons for KISS LUA Script: 

https://www.youtube.com/watch?v=ksF3GtHy5KU

[image]

DieHertz, 2018-06-23:
Honestly I would prefer Enter to be long click on joystick as well, I'll look into it

ps915, 2018-06-23:
Yes, that would be also my Nr 1 but if you hit Enter long you get "telemetry reset" and thats not LUA Script related programming, its OpenTX, right?