betaflight/betaflight-tx-lua-scripts #138: 3.4 Overhaul

Bradw-17, 2018-06-12:
Needs cleaning up for 3.4, especially with release candidates coming out. Additionally, 2 values I've noticed I can no longer change include Dterm lowpass and vtx output power.

mikeller, 2018-06-12:
Yes it does. We were somewhat log jammed on this because the memory issue that were plaguing OpenTX 2.2.1 caused the lua scripts to fail in many cases, making it hard to develop / release updates.

Now that 2.2.2 is out we'll confirm that this has fixed the issues, and then get back onto it.

RipperDrone, 2018-06-26:
2.2.2 (RC) is out, haven't seen mem problems since on Taranis X9D

Bradw-17, 2018-07-08:
2.2.2 RC4 is out, they suggest it will be very similar to the actual release.

RipperDrone, 2018-07-20:
1.0.2 lua script still messed up with BF3.4:

- does not display rate values anymore (leaves fields empty, all other pages display fine n Taranis X9D+)
- still annoying 'retrying...' message whenever trying to save a page... :-/

closed 2018-08-04 by Bradw-17