betaflight/betaflight-tx-lua-scripts #139: Can't save D weight

ctzsnooze, 2018-06-14:
I have the 1.02 Lua scripts on an X9D running 2.2.1

On the page Rates 2/2 I can set and save all parameters *except* D weight.  If I change it and press save, I get Saving.... and then it reverts to the original value.  All other parameters on that screen can be saved without problems.

I'd be grateful for advice on how this might be fixed.

luismiguelsi, 2018-07-01:
i'd be gratefull too, thank you.

mikeller, 2018-07-01:
I think this was fixed in #140. But it's not yet released, you'll have to run latest from https://ci.betaflight.tech/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/lastSuccessfulBuild/artifact/.

luismiguelsi, 2018-07-02:
oh... perfect! thank you!

codecae, 2018-07-13:
Duplicate of #146

closed 2018-07-13 by codecae