betaflight/betaflight-tx-lua-scripts #145: X-Lite Lua : changes require single stick press for each individual value change

ijustwannafly, 2018-07-02:
I am using the most recent lua script with support for the xlite.

In the main otx menus of the tx the joystick long press on options ie model name will allow a fast scroll through all the values.

In lua long pressing up or down to change values results in no change except for one single digit value.

So if I needed to change a value from 10 down to 0 I would have to click the stick 10 times to get there rather than hold the stick.  10 is a small number but imagine changing other values (tpa throttle value or rates for example).  It takes a while.

Also, stick up decreases values while stick down increases values.  

Many thanks.

DieHertz, 2018-07-02:
I also noticed this in Crossfire and fixed it, will find for Betaflight tonight

mheberlein, 2018-07-08:
I was also wondering why up/down decrease/increase inverse to what I'd expect.
As this is also mentioned in the release notes, is there any deeper reason behind it or can it just be flipped?
I know X-lite was just newly added, thanks for that!

DieHertz, 2018-07-08:
Could you try the script from this branch?
https://github.com/DieHertz/betaflight-tx-lua-scripts/tree/bfdev-diehertz-fix-xlite-key-directions

codecae, 2018-07-13:
Based on what is in the OpenTX lua api, there aren't any repeat (REPT) events for XLite.  Consequently, each event must be handled with single button presses.

https://github.com/opentx/opentx/blob/xlite-prod-batch2/radio/src/lua/api_general.cpp#L1409-L1424

hsuq510, 2018-07-18:
pageDown = EVT_PAGEDN_FIRST,
pageUp = EVT_PAGEUP_FIRST or EVT_LEFT_BREAK to inc/dec 10 would be better.

ijustwannafly, 2018-08-18:
I think I got this sorted out.

Here are the changes I made (after many attempts) to get it working on the xlite...at least.

[events.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2299904/events.txt)
[ui.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2299905/ui.txt)

Changes:

Menu save moved to shift long press instead of joystick right long press
Joystick left/right navigates the pages accordingly
Joystick up/down navigates the page lines accordingly
Joystick up/down increments/decrements values accordingly
Joystick long press up/down increments/decrements values accordingly as a scroll
Joystick right long press for save menu removed
Joystick menu long press for save menu removed since the xlite would also popup a telemetry reset window

In ui.lua (ui.txt) the changed lines are 
```
   -- editing value
    elseif currentState == pageStatus.editing then
        if (event == userEvent.release.exit) or (event == userEvent.release.enter) then
            currentState = pageStatus.display
        elseif event == userEvent.press.minus or event == userEvent.longPress.minus or event == userEvent.dial.right then
            incValue(-1)
        elseif event == userEvent.press.plus or event == userEvent.longPress.plus or event == userEvent.dial.left then
            incValue(1)
        end
    end
```

events.lua (events.txt)
```

return 
{
    press = {
        minus = EVT_MINUS_FIRST,
        plus = EVT_PLUS_FIRST,
        pageDown = EVT_PAGEDN_FIRST or EVT_RIGHT_BREAK,
        pageUp = EVT_PAGEUP_FIRST or EVT_LEFT_BREAK
    },
    longPress = {
        menu = EVT_SHIFT_LONG,
	minus = EVT_DOWN_REPT,
	plus = EVT_UP_REPT
    },
    repeatPress = {
        minus = EVT_MINUS_REPT,
        plus = EVT_PLUS_REPT
    },
    release = {
        enter = EVT_ENTER_BREAK,
        exit = EVT_EXIT_BREAK,
        menu = EVT_RIGHT_BREAK,
        minus = EVT_DOWN_BREAK or EVT_MINUS_BREAK,
        plus = EVT_UP_BREAK or EVT_PLUS_BREAK
    },
    dial = {
        left = EVT_ROT_LEFT or EVT_UP_BREAK,
        right = EVT_ROT_RIGHT or EVT_DOWN_BREAK
    }
}
```