betaflight/betaflight-tx-lua-scripts #147: FrSky X-Lite - more intuitive buttons

ps915, 2018-07-09:
I tested the latest LUA Script with OpenTX **2.2.2 RC2**.

Navigation is just fine but the **"save page"** dialog is on **"long right"** button.

Wouldn't it be better (more intuitive) to use the **"Function/Shift Button"** to get to the menu?

Here is a video, how we set the buttons for KISS LUA Script: 

https://www.youtube.com/watch?v=ksF3GtHy5KU

[image]

ps915, 2018-07-14:
Its working and feels better than the one which is realeased now. What you guys think?

hsuq510, 2018-07-17:
also changing a numerical value will imply to move up/down and release the multi function joystick and if you need to increase/decrease a value of 100 digit its a PITA. Will suggest to use a long press or remap the event as in xlite there are new event assigned:

https://github.com/opentx/opentx/blob/2.2/radio/src/lua/api_general.cpp#L1409

Maybe a long press left/right when page is edit mode can increase/decrease the value by 1.

closed 2018-08-23 by stale[bot]