betaflight/betaflight-tx-lua-scripts #148: No Vaules

btmerz, 2018-07-09:
Hi,
I have the script running on my taranis with 2.2.1 and can scroll to all pages witout the memory error. I also receive telemetry from the BF copter and I don't get the "No Telemetry" warning in the script.
So everything seems fine, except that I don't get any values only ---?
What can be the reason for that?

btmerz, 2018-07-09:
I've forgotten to metion that I'm runnunf BF 3.2.4 final release...

mikeller, 2018-07-10:
What RX / RX firmware version are you using?

btmerz, 2018-07-10:
I'm using Version 2.2.1 (cca71808) from 2017-12-27.
Lua and Luac is activated

mikeller, 2018-07-10:
No, the firmware on your RX.

btmerz, 2018-07-10:
I don't know. It's an Frsky XSR and around two years old...
I've never updated it...

mikeller, 2018-07-10:
Then I'd try to update it to the latest version...

btmerz, 2018-07-10:
OK. I will give it a try...

D4Za, 2018-07-13:
i got the same problem, anything new?

wascheck, 2018-07-22:
I have the same problem and I have the XSR receiver on both quads and the using the last version of firmware 170619

wascheck, 2018-07-22:
I have the same problem.
Taranis QX7 Open TX Ver. 2.2.2 RC1
FRSky XSR RX with Ver. 170619
LUA Script Ver. 1.02

I've downgraded FC firmware to BF 3.4.0 RC5 and worked ok.
There is a bug on the Final Version.

The problem is that the Telemetry is ok but the Lua Script doesn't show the value of the PID..etc.

mikeller, 2018-07-23:
@wascheck: Fixed in #6382, will be in 3.4.1.

closed 2018-08-29 by stale[bot]