betaflight/betaflight-tx-lua-scripts #151: BF 3.4 + 3.5 Nightly, R-XSR (inverted) on FPORT - runs like crap

RipperDrone, 2018-07-24:
All kinds of glitches:

- sensor lost messages quite freqently on Taranis X9D+(even though I have no ESC telemetry enabled, no telemetry sensors like ESC, fuel, AccX/Y/Z, GPS set up
- some lua pages loading values, others don't
- some values saving, others don't
- generally annoying 'retrying' message every time, even though sometimes values are actually saved

Taranis FW is 2.2.2 RC5, lua scripts running somewhat more stable on BF3.4 than on 3.5... :-/

mikeller, 2018-07-24:
What firmware do you have on your RX?

Also, providing information about the versions, and a config `diff`, like you were asked to do when opening this pull request, will greatly improve the chances of anything being done about your rant.

RipperDrone, 2018-07-25:
No intention to make it understood as a rant / see smilie in title.

I didn't provide all the details since I thought it might be a general issue (also mentioned in some other issue reports before) that the latest lua script still does not cover all the new timings of BFf3.4+ regarding FPORT protocol.

Anyway, here's the details you asked for:
- RX Firmware version is 180228, latest stable from FrSky webpage
- BF 3.4 stable, BF 3.5 Nightly #1010 tested
- diff link: https://drive.google.com/open?id=1PD-H_9OGwnP7vYVieSq3KS48oPFjYnU_

mikeller, 2018-07-26:
@RipperDrone: For some reason I get a 404 when trying to open your diff. Can you try 3.5.0-RC1? betaflight/betaflight#6397 disables sending of the per-ESC telemetry data, reducing the bandwidth going through telemetry.

RipperDrone, 2018-07-26:
Link updated. Will test 3.5RC1 next...

RipperDrone, 2018-07-26:
mixed results:

Quad1 = OBF4: All telemetry fine with BF 3.4 stable
Quad2 = DYSF4PRO: BF3.4 = 2 missing tele pages Rates and motor PWM. BF3.5RC1: No telemetry pages at all with lua script.

RipperDrone, 2018-07-26:
Flashed Quad2 back to BF3.4 - again script working but for 2 pages (no values shown, others are fine)...

Guidus93, 2018-07-27:
Also here, no values on Lua with 3. 5,it was working with 3.3 and 3.4 TOO. RXSR with stock eu firmware and Fortini 
Is there something I should check on cli parameters?

RipperDrone, 2018-07-27:
must have to do with altered tele frames + timings...?

klutvott123, 2018-07-27:
With 3.5.0 RC1 i stopped getting "sensor lost" all the time, but as soon as i switch to the telemetry page with the lua script i get "sensor lost" after a few seconds. I don't get any values on any pages of the lua script either. If i don't switch to the lua script everything is fine and telemetry works perfectly. 

Tried 3.3.3 again just to test and it shows values on most of the lua pages( not PWM and VTX). I can change and save values, but if i do it too much it's "sensor lost" again. With 3.3.3 i have to wait for a few minutes for the sensor to come back. Rebooting FC/transmitter does nothing. "sensor lost" seconds after switching on. 

I'm starting to suspect that the issue is with the R-XSR as it seems like most people with these issues are using it. Those using other receivers had issues with ESC telemetry enabled. 
Also @mikeller mentioned in #120 that the R-XSR has a 8051 microcontroller instead of the stm32 that the other frsky receivers are using and suggests that it might not be capable of handling the same number of sensors. 

I might get a X4RSB to see if there's any difference. 

Using SPRACINGF3 clone, R-XSR  f.port 180615 beta firmware(tried all firmwares),
Taranis q x7 with opentx 2.2.2 RC5.

RipperDrone, 2018-07-27:
R-XSR FPORT was spot on with BF3.4 (had all the probs you described before thqat) on many FCs like OBF4, OBF4SD. On DYSF4PRO,  2 telemetry pages over lua were not displaying values, all the others are ok. So since firmware version 180228, the R-XSR seemed to work almost by 100%. But then, with BF3.5RC1, none of the values / pages was working anymore.

It's true the R-XSR has a different MCU, and @mikeller has done a great job in adjusting frames frequency, timing and so forth. But it still seems there is some timing glitch in BF3.5 now...

mikeller, 2018-07-27:
@RipperDrone: Unfortunately, given we have got SmartPort and FPort, a number of different RX, all with a number of firmware versions, 'making it work' is more like 'herding cats'...

Can you try the 180615 version for r-XSR (under 'Beta firmware for FPort)?

RipperDrone, 2018-07-28:
Yes, I know. Will update the OBF4 where RX is easier to access. On my Airdancer with the DYSF4Pro, I would have to tear a lot of soldering and shrinks apart since the RX is buried deep underneath the center FC section which is tied down / constrained by short ESC cables and shrik wraps around the arms. Extremely painful, trying to avoid this. When on a promising track, I can test with other rigs, though...

It seems that the worst combination right now is the DYSF4PRO / R-XSR FPORT, where I don't get ANY of the lua telemetry pages filled with readout values. This rig has BLHeli32 individual ESCs which issue telemetry data. I have tied those 4 telemetry pins together and attached them to one UART.

RipperDrone, 2018-07-28:
Did some testing now, rather frustrating than enlightening... :-)

- OBF4 quad didn't show any changes after flashing latest FrSky FPORT Beta (180615). I then somehow must have kiled the gyro of the HGLRC stack (really bad quality soldering they do), so I had to replace the FC and solder everything new. Connected SBUS + separate telemetry this time. Now BF 3.4 and SBUS latest FrSky firmware on R-XSR working well. Cutting FPORT out of the equation did help here. Of course, it doesn't help the team here to fix FPORT, though...
- other FC DYSF4PRO (which I can't access the RX easily w/o destroying the whole build basically :-(): I tried BF3.4 and 3.5 with follwing swaps: Telemetry feature ON/OFF, ESC sensor feature ON/OFF, Port settings including ESC: ON/OFF, serial settings inverted / halfduplex all combinations. In any of these cases, lua script 1.0.2 works well on BF3.4 except the 2 pages rates and motor PWM. On BF3.5 #1033 NO page is being displayed with any values, completely dysfunctional. There must be something fundamentally changed btw BF3.4 and 3.5 re lua telemetry redout timings / frames...

RipperDrone, 2018-07-28:
to try flash latest R-XSR FPORT beta firmware w/o rebuilding the whole quad: Is there a way to flash the RX over serial passthrough in the meantime so I could stay wired up as it is?

RipperDrone, 2018-07-28:
found something: Rate page hasbeen empty in BF3.3 as well under certain circumstances... https://github.com/betaflight/betaflight-tx-lua-scripts/issues/117

RipperDrone, 2018-07-28:
Could it be PR https://github.com/betaflight/betaflight/pull/6300 causing the difference BF3.4 <-> 3.5 maybe?

klutvott123, 2018-07-28:
Changing FPORT_MIN_TELEMETRY_RESPONSE_DELAY_US to 100 in fport.c got rid of the "sensor lost" for me when opening the lua script. Still no values on any pages. Then i increased FPORT_TELEMETRY_MAX_CONSECUTIVE_TELEMETRY_FRAMES to 10 and now i get values on all pages except PWM and VTX(don't know if i'm supposed to have vtx values since i don't have a vtx.). 
Still a little slow. Takes a few seconds to load pages but i can change values and save them. Just flew 4 packs and did a little pid tuning with no problems other than one quick "sensor lost". 
Works for my setup but might not for other setups. I can see how getting this to work for every setup is like "herding cats". 

Also compiled 3.5.0 with fport.c from 3.3 and that also works(still no PWM and VTX). I think it's because 3.3 didn't have the restriction on consecutive telemetry frames at all.

mikeller, 2018-07-28:
@klutvott123: `FPORT_MIN_TELEMETRY_RESPONSE_DELAY_US` is something that was directly specified by FrSky, so making it shorter will potentially lead to problems with future versions of the RX firmware, since that's the spec they make it for.

`FPORT_TELEMETRY_MAX_CONSECUTIVE_TELEMETRY_FRAMES` had to be introduced to prevent buffer overruns on r-XSR. This is superseded by the change introduced in betaflight/betaflight#6098 that allows the RX to signal to the flight controller when it's ready to get another telemetry frame. But this requires RX firmware support. Currently the r-XSR 'Beta firmware for FPort' supports this, and there is XSR firmware in the works for it.

klutvott123, 2018-07-28:
@mikeller Thanks for explaining. I just reflashed my R-XSR with the latest 180615 beta fport firmware just to be sure and now it's working without changing FPORT_MIN_TELEMETRY_RESPONSE_DELAY_US and FPORT_TELEMETRY_MAX_CONSECUTIVE_TELEMETRY_FRAMES. I was sure i was on 180615 but must have had 180228 instead(have been switching between the two to test). 

The PWM and VTX pages are still empty. Do you know what might be causing this?

RipperDrone, 2018-07-29:
Weird thought crossed my mind: Does anyone of you who is having the arms drop problem have pid_at_min_throttle switched off? It's off on all my quads since I prefer to have the quad sitting idle on launch, no slow I term rpm windups whatsoever, no reaction to stick inputs unless I raise throttle.

I understand that this switch shouldn't matter since once a certain throttle threshold is surpassed, PID control should engage and stay on from then, so even hard throttle cuts back to min throttle should be PID covered. Confirmed? (I will try to swap the setting of this option, anyway... 🙂)

RRAlex, 2018-08-10:
Same issue here on OpenTX 2.2.2 and BF 3.4.1 with F.Port 180228:

Lost telemetry & slow update (though I can't compare, this is my first build...)

RRAlex, 2018-08-14:
@klutvott123 Oh, I see, didn't realize there was a beta, will try that and give some feedbacks, thanks!

atomiix, 2018-08-15:
Same issue here on OpenTX 2.2.2, BF 3.5.0 and X4RSB with F.Port 171114 (last nonEU version).
Some "sensor lost", slow update & empty PWM & VTX.

klutvott123, 2018-08-15:
In https://github.com/betaflight/betaflight/blob/master/src/main/interface/msp.c line 1240:

`    case MSP_ADVANCED_CONFIG:
        sbufWriteU8(dst, gyroConfig()->gyro_sync_denom);
        sbufWriteU8(dst, pidConfig()->pid_process_denom);
        sbufWriteU8(dst, motorConfig()->dev.useUnsyncedPwm);
        sbufWriteU8(dst, motorConfig()->dev.motorPwmProtocol);
        sbufWriteU16(dst, motorConfig()->dev.motorPwmRate);
        sbufWriteU16(dst, motorConfig()->digitalIdleOffsetValue);
        sbufWriteU8(dst, gyroConfig()->gyro_use_32khz);
        sbufWriteU8(dst, motorConfig()->dev.motorPwmInversion);
        sbufWriteU8(dst, gyroConfig()->gyro_to_use);
        sbufWriteU8(dst, gyroConfig()->gyro_high_fsr);
        sbufWriteU8(dst, gyroConfig()->gyroMovementCalibrationThreshold);
        sbufWriteU16(dst, gyroConfig()->gyroCalibrationDuration);
        sbufWriteU16(dst, gyroConfig()->gyro_offset_yaw);
sbufWriteU8(dst, gyroConfig()->checkOverflow);`

The bottom 6 lines are new from 3.4. I tried commenting them out just to test and that got the PWM working again. I can't see why it's not working with the above code. The FC just sends all those bytes and the lua script uses the first 9.  I tried sending the MSP_ADVANCED_CONFIG command in a terminal to see what exactly comes out but didn't have any luck with that.

atomiix, 2018-08-15:
@klutvott123 I can confirm, I went further and tried to comment each line one by one and it's the line `sbufWriteU16(dst, gyroConfig()->gyroCalibrationDuration);` which cause the issue.

So, I did `set gyro_calib_duration = 128` in the CLI and the PWM page was working again !

klutvott123, 2018-08-15:
@atomiix I just set it to 128 too and it's working now. Thanks!

This has to be some kind of bug because i have something like this happening on the PID page too. Some combinations of pid gains just makes the page blank. Right now if i set the pitch I to 60 and roll I to 55 it stops working. Earlier i had a case where setting yaw I to 55 it stopped working. I could set it to 54 or 56 and all other values but not 55. 

Same thing also happens when setting the d-term notch filter center frequency to 125 and the cutoff to 75.

atomiix, 2018-08-15:
@klutvott123 I had exactly the same problem when I was tuning my PIDs, that's what led me to try another value for `gyro_calib_duration`

atomiix, 2018-08-16:
It seems that under certain circumstances, F.Port frames are dropping packets.

I got the line `elseif bit32.band(mspRemoteSeq + 1, 0x0F) ~= seq then` of the file [src/SCRIPTS/BF/MSP/common.lua](https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/MSP/common.lua#L105) (line105) being called with the page PWM (which means packet drop).
It occures only on F.Port, I tried back with SBUS/SmartPort and it's working well.

But I'm not competent enough to investigate more on the F.Port protocol 😅

mikeller, 2018-08-16:
@atomiix: There are a number of places where telemetry frames can be dropped:
- due to having only a 2 bit sequence number on the radio link, any loss of 3 or more frames will result in a data loss on FPort and SmartPort;
- in FPort, the initial implementation had the flight controller send more telemetry data to the RX than the RX could send over the air, causing queue overruns. This has been fixed in the latest Betaflight versions, but the fix so far only exists for R-XSR on the RX side ('Beta' firmware).

klutvott123, 2018-08-17:
@mikeller I just read the f.port protocol documentation and did some experimenting. If i set any value on any lua script page to 125 or 126 it stops working. 
126 is the FPORT_FRAME_MARKER 0x7E and 125 is the FPORT_ESCAPE_CHAR 0x7D in https://github.com/betaflight/betaflight/blob/master/src/main/rx/fport.c. This makes me believe the f.port bytestuffing isn't properly implemented somewhere. 

https://github.com/betaflight/betaflight/blob/98616d469012e5287f35c9302bbea70d1e8b920d/src/main/telemetry/smartport.c#L274 This looks right to me so i assume the f.port frame is correct when sent to the receiver. Is there a chance that it's not properly "unstuffed" at the receiver side and not getting sent to the transmitter at all when the data contains 125 or 126?

Using betaflight 3.5, R-XSR with 180615 beta f.port firmware.

klutvott123, 2018-08-17:
Did some more testing. I set my PIDs so that the PID page stopped working. I wrote a python script that sends the MSP_PID command to the FC and prints the received data. The combination of pid gains that makes it stop working has a CRC of 125(FPORT_ESCAPE_CHAR 0x7D). Now i really think this is a bytestuffing issue.

mikeller, 2018-08-18:
Good find, @klutvott123!

On the flight controller side, FPort uses the same byte stuffing that SmartPort uses: https://github.com/betaflight/betaflight/blob/98616d469012e5287f35c9302bbea70d1e8b920d/src/main/rx/fport.c#L237-L239

But where I can see a potential difference is that, because with FPort, the frame delimiters are used for RC control and telemetry frames on the same connection, any occurrence of a stray delimiter would have way worse consequences than on SmartPort, where the RC control is unaffected.

My suspicion is that the lua API that we are using to inject the MSP data expects the data to be byte stuffed when it is sent, or is simply buggy and does not do the byte stuffing correctly.

klutvott123, 2018-08-19:
@mikeller Some more details which i should have added: 
When setting a value in the lua script to 125 or 126 and saving, it get's sent correctly to the FC. It's when the script tries to update the page again that it fails and all fields are blank. When i check the configurator i can see the value that i just set in the script. So it's working when sending data from transmitter but not when receiving.

You may be right about the lua API. I just had a quick look at some of the opentx s.port code and it looks like it's unstuffed correctly. Will investigate some more.

klutvott123, 2018-08-24:
@mikeller So i have done some testing and made some observations. 

I soldered a UART module directly to the f.port pad on my FC and printed the data while the lua script was sending the pid command 112. I set my yaw I gain to 125 so that the frame had to be bytestuffed. The data for the pids is split up into four frames, is bytestuffed correctly and the CRCs are correct so no problems there. Just had to make sure. 

Then i modified the function that processes frsky telemetry data to check for the byte sequence i'm interested in and print that frame directly to the transmitter lcd, bypassing lua completely. Works fine for the frames that aren't bytestuffed but for the bytestuffed frames i get nothing. This makes me think that the bytestuffed frames aren't received at all. 

I had my transmitter turned on for a while and suddenly got a "sensor lost". Looked at the telemetry screen and saw the fuel sensor blinking with the value 124. It was gone until the fuel sensor reached 127. So the sensor got lost when the fuel sensor frames were bytestuffed. 

I also think i know why the lua pages are updating so slowly. For uplink data frames the CRC is calculated from all the bytes in the frame. APPID bytes included where the first byte is the frameID that isn't the same everytime. Sometimes the frameID causes the CRC to become 125 or 126 and then the frame has to be bytestuffed. The lua script then has to request the data again and again until none of the data or CRCs of the frames are bytestuffed. 

Can also confirm that s.port works correctly.

RipperDrone, 2018-08-25:
Wow, @klutvott123 - you have finally found the explanation for what has been plaguing so many users for a long time. Wonderful work, thank you! Now it seems about how to do bytestuffing / destuffing the correct way, without the influence of channel values 125, 126...

Will this have to be done with FrSky devs, or is it something on the BF code side?

bhuism, 2018-08-25:
@klutvott123 awesome

klutvott123, 2018-08-25:
@RipperDrone I'm not 100% sure yet. I'll have to do some more investigating. 

Just did some testing with https://github.com/betaflight/betaflight/blob/9ed2914489fc12e60bcb1d8af1bdb65677518aa2/src/main/telemetry/smartport.c#L274 where i commented out the if else statement that takes care of bytestuffing on the FC side. Then i added serialWrite(port, c); below so that no bytestuffing is done on the data sent from FC to RX. 
Still using my pids for testing this and when i set any value to 126 it works. I get the correct sequence of bytes in opentx. The byte that comes after the pid value that is set to 126 is also shown correctly. Still doesn't work with 125. 

@mikeller is there a chance that the f.port firmware expects to do the bytestuffing itself? The length of an uplink dataframe is 8 bytes according to the documentation. When bytestuffing is done on the FC side the frame length exceeds those 8 bytes.

RipperDrone, 2018-08-25:
@mikeller Is there anybody from FrSky active on BF GitHub so we can tie that rep/dev into the coding himself/herself? Reverse engineering the protocol details by @klutvott123 might be a tedious task to do...

RipperDrone, 2018-08-25:
Sry for being too unexperienced to really comment in a knowledgeable and helpful way, but what I understood so far is there seems to be a mismatch btw sender and receiver regarding bytestuffing logics. Actual values of 125 (0x7D) seem to be interpreted wrongly as a bytestuffing yes/no flag, so the value itself is not transmitted properly anymore. I skimmed the telemetry.c code for occurrences of the FSS_DLE (=0x7D) flag and found several places where the value itself is still transmitted, but e.g. in this one, it isn't:
[image]

Anyway, we desperately need a FrSky dev here to look into the receiver side / their protocol details to resolve it quickly... :-/

klutvott123, 2018-08-25:
@RipperDrone That code is for receiving s.port/f.port data and it works correctly. In betaflight f.port is implemented as it says it should be in the f.port documentation. The only thing is that in the documentation it says bytestuffing should be done by "sender". Sender could be interpreted both as FC and RX as they are both senders. 

Heres some example data from my FC f.port pad:
```
126 8 1 48 53 0 112 112 0 0 176 126 <-- MSP_PID 112 from lua script

8 129 50 23 15 46 45 25 50 119         <-- First frame sent by FC containing PIDs

126 25 0 228 0 31 248 196 7 12 96 0 3 24 0 0 0 0 0 0 0 0 0 0 0 0 84 66 126

126 8 1 0 0 0 0 0 0 0 246 126

126 25 0 228 0 31 248 196 7 12 96 0 3 24 0 0 0 0 0 0 0 0 0 0 0 0 84 66 126

126 8 1 16 0 0 0 0 0 0 230 126

8 129 50 8 51 27 40 125 93 0 72   <--Second frame sent by FC. 125 is yaw I gain. Here it's bytestuffed as 125 93

```

klutvott123, 2018-08-25:
I think i can confirm that there's bytestuffing going on in frskys f.port firmware. 
Set my yaw I gain to 126. The databytes sent from FC to RX are:
```
51 27 40 126
```
The databytes that comes out in opentx before any processing is:
```
51 27 40 125 94
```
This is with the codechanges i described above that disables any bytestuffing of data to be sent in betaflight. Still no idea why it doesn't do it when the frame contains 125.

bhuism, 2018-08-25:
I love it, feel a work around is around the corner, no pun intended 😁

mikeller, 2018-08-26:
@klutvott123: I did a couple of tests with r-XSR and XSR, and they both seem to be handling this in the same way, which is weird since they do not share the codebase, making it unlikely to be an unintentional bug.

Can you run the tests above to see what 125 comes out as?

@RipperDrone: Unfortunately most of the FrSky engineers do not speak English, making communicating with them only possible via proxy. But I am working my contacts to try and get them to look into this, and comment on it. Also, there are different teams responsible for the firmwares for XSR, r-XSR, and R9, making this even harder.

mikeller, 2018-08-26:
Update: There is a difference: Disabling byte-stuffing will not fix 126 on XSR. And I am talking to FrSky, they will be looking into it.

mikeller, 2018-08-26:
@klutvott123: You seem to be set up to do in / out comparisons. Can I get you to provide some in / out samples for byte-stuffed versions of 125 and 126 (i.e. `125 93` and `125 94`)? According to FrSky, the specification has to be interpreted to do the byte-stuffing in the same way as it is done in SmartPort, i.e. the flight controller has to do the bytes-stuffing.

klutvott123, 2018-08-26:
@mikeller Ok, so i changed the smartPortSendByte function back so that it bytestuffs as usual. The thing is that when data from the FC has 125 or 126 in it, it doesn't show up at the opentx end at all so there's nothing to compare to. It turns out that enabling telemetry logging when compiling opentx makes it log everything. Couldn't find the frames containing 125 or 126 there either. 

Here's an example from the log: 
```
2018-08-26,12:27:29.870: 7E 00 32 16 0F 2E 2D 19 32 02   <-- First frame after sending MSP_PID
2018-08-26,12:27:29.880: 7E 00 32 08 35 32 4B 28 00 EA   <-- Third frame
2018-08-26,12:27:29.900: 7E 00 32 09 00 4D 00 00 00 77   <--Fourth frame

```
Second frame contains 125 or 126 and is bytestuffed in the FC. This pattern repeats itself through the log. 

Another example from the log: 

```
2018-08-26,12:27:32.150: 7E 00 32 02 00 4D 00 00 00 7D 5E
```
This frame contains the CRC for the MSP frame itself and its 0x4D. The CRC for the f.port frame at the end is 0x7E and has been bytestuffed, BUT it has to have been bytestuffed by the RX since the CRC would be different for the frame coming from the FC since instead of 0x00 at the beginning it would have been 0x08 and 0x81 and the CRC would have been calculated from that. So the RX takes the data from the FC, extracts it and repacks it. 

One last example:
```
2018-08-26,12:26:48.060: 7E 00 32 01 00 4F 00 00 00 7D 5D
```
This also contains CRC for MSP frame. The RX has calculated a new f.port CRC of 125 and bytestuffed it. 

Even if the frames from the FC containing 126 were to be passed on to opentx they would have been double stuffed by the RX. At least for the R-XSR since your testing shows that it doesn't do it on the XSR. 

Also thank you for contacting frsky about this!

mikeller, 2018-08-26:
Thanks for the testing @klutvott123. From the look of it, the RX discards all frames that contain an 125 - this is definitely a serious problem.
The byte-stuffing of the 126 in itself is not a problem, since 126 should never be received on the RX side. Also, this can be explained with the fact that the byte-stuffing is only used on the serial links on both sides, but not on the radio link (i.e. an 126 is always sent as an 126 over the radio).

bhuism, 2018-08-27:
I will test later today, thanks @mikeller

RipperDrone, 2018-08-27:
@mikeller Did a thorough test on Ummagawd / Omnibus F4SD / R-XSR / BF 4.0 #1082 / FPORT / Taranis X9D+ OTX2.2

FrSky FPORT firmware 180615_Beta:
- When I set a PID value of 125 or 126, it does not read back these values on the RC - shows empty fields only
- Telemetry page Betaflight / PWM completely empty, LUA script on Taranis never successfully reads any content

FrSky FPORT firmware 180827_Beta (which you provided from FrSky):
- both 125 and 126 values are correctly read back :-)
- Betaflight / PWM page readout successful! :-)

Still not resolved (but minor glitch): Whenever I write back altered values over Lua Script, the Taranis screen says 'Saving', then 'Retrying'. Values are still stored correctly.

Summary: BIG WOW! We finally got this nasty bug resolved - R-XSR FPORT is back in the game! Loving it, great team effort and super smart finding by @klutvott123 👍

Billj747, 2018-08-27:
@RipperDrone - FrSky FPORT firmware 180827_Beta solves the LUA script / "Sensor Lost" problem?

Where can I find that software version?  All I see under the frsky 'beta' firmware page is 180615.

mikeller, 2018-08-28:
@Billj747: Here: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-416183224

mikeller, 2018-08-28:
@RipperDrone: I suspect that the timing used in the lua script is a bit too tight - we'll probably have to extend the time until we show 'retrying'...

RipperDrone, 2018-08-28:
Nice, this is cream on top 👍 Happy to test when needed.

bhuism, 2018-08-28:
@mikeller, @klutvott123, Just flown 4 packs with xsr-r on FPORT firmware LBT180827_Beta, BF 3.5.0 on kakute f7, OpenTX 2.2.2 on an X9D, had **0** 'sensor lost' warnings! very happy with this fix and collaboration, thanks all!

mikeller, 2018-08-28:
@bhuism: Glad to hear that this seems to be fixing these sporadic and hard to pinpoint 'sensor lost' warnings as well - my suspicion is that they occurred whenever a sensor value happened to contain a 125 or 126.

klutvott123, 2018-08-30:
Extending the save timeout makes no difference. I extended it to 10 seconds and it just "saves" for 10 seconds. Then "retrying" every time. I think the data is lost somewhere after being sent. When loading pages there are often frames lost. Sometimes one frame and sometimes all frames(this might also be that the request itself is lost).

mikeller, 2018-09-01:
Ok, then we probably need to add multiple retries before we show 'retrying', or just accept that it won't work reliably first time round in most cases - the way telemetry data is framed for transport makes the link very unreliable for our purposes.

RipperDrone, 2018-09-01:
It's so annoying, plus confusing for beginners. I'd go with 3x retrying w/o throwing the warning, only once it has to retry 4+ times, throw the message.

Alternatively, it could be a certain timeout. Guess implementing a counter is easier...

klutvott123, 2018-09-01:
I just made a discovery. Most of the time the values will actually be saved on the first try, but it goes through all the retries anyway for some reason. I tried increasing the number of retries and it just goes through all of them it seems. 

If anyone has the time, try this. Set this line: https://github.com/betaflight/betaflight-tx-lua-scripts/blob/523fa04df1075639cd0fb4c9531171475fddae20/src/SCRIPTS/BF/protocols.lua#L12
to 0.

This allows it to save only 1 time. Will investigate more later.

SomeguyinPDX, 2018-09-01:
Good morning to all.  If this is the wrong place please correct me.   I have been following this thread hoping it may have some answers to my R-XSR Lua script issues.   I'm am a brand new convert from Spektrum & a computer idiot so 95% of this thread is Greek to me.   I gave up trying to get Fport to work with the XRotor F4 45A ESC combo so I am currently wired with the non-inverted S.port pad from the R-XSR.    Lua scripts are showing telemetry data with the occasional "sensor lost" warning but no changes to the PIDS and VTX actually take place on the FC even though the X9D shows the change in the Lua script page until power is cycled then the telemetry updates and shows the original settings. Although if I use the BF OSD/stick menu I can make the same PID/VTX changes and they actually change.

[105 more replies]

closed 2018-09-05 by RipperDrone