betaflight/betaflight-tx-lua-scripts #152: FEATURE REQ: CLI commands

mickkn, 2018-07-30:
Would it be possible to actually somehow make a script which would be able to type cli commands to the FC? via the Taranis?

And maybe also by using GET to read the current value?

Bradw-17, 2018-07-30:
using buttons or click wheel would take a ridiculous amount of time, just saying.

mickkn, 2018-07-30:
Indeed, but I guess a function to scroll through all the commands would take up a lot of memory :) but would also be nice. 

This would just be a last resort in the field :)

FieserKiller, 2018-07-30:
I'd love that function. Sure, typing commands would take ridiculously long - but we could control everything this way. most importantly features which are not yet supported by the scripts gui part.

mikeller, 2018-07-31:
Hmm, would be interesting if it was possible to forward the MSP / CLI data stream from the flight controller to the USB port on the TX, and then on to a mobile phone connected over USB OTG...

closed 2018-09-06 by stale[bot]