betaflight/betaflight-tx-lua-scripts #157: Empty PWM page

simogere, 2018-08-21:
FC: Airbot Fireworks V2 (with Betaflight 3.4.1)
RX: FrSky R9 Mini (with last firmware)
Radio: FrSky Taranis Q X7 (with OpenTx 2.2.2)

Lua and luac enabled. I can manage all the screens of the lua script excepting the PWM page that is empty.
With last Butterflight release, the PWM page works.

mikeller, 2018-08-21:
@simogere: Can you try again with Betaflight 3.5.0?

simogere, 2018-08-22:
Tried, but no values again.
If possible let's focus to 3.4. 
3.5 is not good enought for my quads. Hope you understand. 
Is there any code patch I could try?

mikeller, 2018-08-22:
No. And development on 3.4 has finished, so there will be no more fixes for it. We have only so much free time to invest in developing this, I hope you understand.

mikeller, 2018-08-28:
@simogere: Yes please - 3.5 is the current release, and if there is a bug in it it will be fixed.

simogere, 2018-08-28:
Ok, but which release should I use? 1.0.2?

mikeller, 2018-08-28:
Yes, 1.0.2 should work.

RipperDrone, 2018-09-01:
empty PWM page on FPORT+R-XSR was fixed with firmware 1800827_Beta from FrSky, however this version had timing issues on Omnibus F4 target (EMAX F4 Magnum Mini v2). Therefore it's a catch 22!

It seems that with the older firmware 180615_Beta, PWM page is not shown but 'sensor lost' issue shows up. Older firmware had issues with bytestuffing, see https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151

mikeller, 2018-09-02:
@RipperDrone:

Can you supply more information about the timing issue please, so we can look into fixing it?

RipperDrone, 2018-09-02:
@mikeller - @klutvott123 is working on it, I'm in discussion with him re timing in Lua script, and I think we just started narrowing it down. However, my knowledge is still too limited to go into full depth with timers used by Lua. It seems that timings created by getTime() and adding some timeout periods are not properly obeyed by the watchdog algorithm, however the algorithm itself looks correct... <scratch head>. See #151 ...

RipperDrone, 2018-09-02:
@mikeller - I was just able to narrow it down to a point that the new 180827 firmware seems to be running fine on an Omibus F4SD, including showing the PWM page properly. I think somebody had the suspicion that bad bytestuffing in the previous firmwares <=180615 might have lead to the whole PWM page not being shown once there is a 125 or 126 data value entry on it somewhere (not sure why there should be such a value, though...)

On an Omnibus F4 (EMAX MAgnum F4 v2), though, the 180827 firmware does not work properly. But this time it's not about showing or not showing the PWM page, now it's timing issues with FPORT telemetry:

When I watch the asterisks on my Taranis indicating sensor values coming in, there are a few fast blinking values like RSSI and RxBat. Other values seem to get updates ore slowly. When continuing to run the radio connection, updates come in more and more slowly, up to a point that the RC times out and throws a 'sensor lost' message. Sometimes it recovers, but after ~1min the data connection seems so fragmented that for some sensors, they don't come through at all anymore and blink on the RC's telemetry screen (=dead). CPU usage (which I suspected first) is at 8% only, I tried both 8k/8k and 4k/4k, no change. I deleted all but 6 sensors on the Taranis, but still the same phaenomenon persists.

Particularly VFAS, A4, Hdg and Accx/y/z seem to be dying first, RSSI+RxBat seem to survive a little longer. 

I have connected the FPORT pin of the R-XSR to the B10/TX3 pin of the MCU since there were no more free external pins to access. TX6 is hard connected to the backpack VTX for SmartAudio, UART1 is free - supposed for SBUS/SPEKTRUM connections. So unfortunately I cannot easily rewire the connections to try out another free UART on this stack.

However, once flashing the R-XSR back to 180615 firmware, all timing issues are gone - frequent updates on all sensors whatsoever, no 'sensor lost' messages anymore. Only little drawback is the missing entries/values on the Lua PWM page.

What else could I do to help diagnose...?

mikeller, 2018-09-02:
@RipperDrone: What firmware version are you running on the OMNIBUSF4 that is showing this problem?

RipperDrone, 2018-09-02:
4.0 #1091

mikeller, 2018-09-02:
@RipperDrone: Ok, this is odd - I can't quite see what's going wrong there.

RipperDrone, 2018-09-02:
@mikeller yeah, strange indeed. New R-XSR FW seems to be an improvement on most FCs, so maybe this is a very special case. I will go with the old FW for now, accepting the missing PWM Lua page. Over time, we'll see whether or not there are other reports about timing issues on some targets...

@simogere How is progress on your end?

klutvott123, 2018-09-02:
@RipperDrone do this: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-413305190

Still any combination of values that makes the CRC 125 or 126 will make it stop working again.

fanderko, 2018-09-21:
Hi, the same problem here. 
FC: Omnibus F4 V3 PRO OSD (omnibusf4sd) betaflight 3.5.1
RX: FrSky R9 Mini
TX: FrSky Taranis Q X7 (with OpenTx 2.2.2)
Only PWM page is blank.

chadmbrown, 2018-09-27:
Hi, the same problem here.
FC: Kakute F4v2 betaflight 3.5.1
RX: R-XSR 
FPORT: R-XSR_FPORT_FCC180615_Beta
TX: FrSky Taranis Q X7S (with OpenTx 2.2.2)

Only PWM page is blank. Everything else works great.

klutvott123, 2018-09-27:
@chadmbrown it's a rx firmware issue. Try the new 180827 beta firmware from frskys website. 

@fanderko rx firmware issue for you too. Frsky haven't released a fix for the R9 mini yet. "set gyro_calib_duration = 128" in CLI should bring the PWM page back, but doesn't fix the issue.

chadmbrown, 2018-09-27:
Thanks. That fixed it. Didn’t notice the second beta directory. I thought I had the latest.

Chad

closed 2018-11-03 by stale[bot]