betaflight/betaflight-tx-lua-scripts #158: Anti gravity gain value deviation

dizzeerazcal, 2018-08-23:
Changing AG gain will result in a value of setpoint - 0,0001 upon saving page. E.g. if AG gain is set to 5,00, result will be 4,9999 and so on.

klutvott123, 2018-08-30:
@dizzeerazcal Yeah, i'm experiencing the same. Looks like there's some rounding issues going on in https://github.com/betaflight/betaflight-tx-lua-scripts/blob/3af93627080151e4dbe157b71b6a03b8a5193480/src/SCRIPTS/BF/ui.lua#L231

I think the result after incrementing the value displayed on the screen isn't exact. That result is then scaled up and stored in the actual page values. 

Try changing the function above to:

```
local function incValue(inc)
    local f = Page.fields[currentLine]
    local idx = f.i or currentLine
    local scale = (f.scale or 1)
    f.value = clipValue(f.value + ((inc*(f.mult or 1))/scale), (f.min/scale) or 0, (f.max/scale) or 255)
    f.value = math.floor((f.value*scale)/(f.mult or 1) + 0.5)/(scale/(f.mult or 1))   <--added this line
    for idx=1, #(f.vals) do
        Page.values[f.vals[idx]] = bit32.rshift(f.value * scale, (idx-1)*8)
    end
    if f.upd and Page.values then
        f.upd(Page)
    end
end

```

This just rounds the value after being incremented. Then the rounded value is stored in the page values. 
It also fixes the case where small values like throttle expo is shown as something like 0.0999999999.

@mikeller or other reviewers: Is there anything wrong with doing it like this or can i make a pull request for it?

mikeller, 2018-08-30:
@klutvott123: Looks good to me. Damn float-only systems. :stuck_out_tongue:

closed 2018-10-06 by stale[bot]