betaflight/betaflight-tx-lua-scripts #160: Problem with VTX channels

lorenwey, 2018-08-23:
Hi,
When I try to change the vtx channels, all channels works ok, except channel R2 and R3. It always says "saving..." but don't works.

I have a TBS PRO Race and I can change without any problem with OSD

paulo-serrao, 2018-09-11:
Check the last comments here: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151

mikeller, 2018-09-13:
@lorenwey: If you are using an r-XSR with FPort, can you please try the r-XSR firmware in https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-416183224 and report back if this fixes it?

cexshun, 2018-10-13:
Same issue here using sbus. CL Racing F4s on BF 2.5.1. Taranis X9D+ with Opentx 2.2.2 final. TBS Unify Race Pro.

So this is a confirmed Frsky firmware problem? The issue is not big enough to warrant resoldering and flashing the receiver to fport. Everything works through the OSD programming, so I'm using that for now.

paulo-serrao, 2018-10-13:
The beta firmware mentioned previously solves the issue.

cexshun, 2018-10-19:
Fixes the r-xsr. However, problem still exists on the r9mm receiver which doesn't have a firmware fix that I can find.