betaflight/betaflight-tx-lua-scripts #162: VTX adjustment Vortex 150 script crash

malelo1960, 2018-08-26:
If i go into the vtx settings on my taranis x9d plus and want to change parameters the script crash immediately.
An error window shows "Script syntax error BF/X9/vtx.lua:192 attempt to index field '?' (a nil value)
I have to restart the taranis to get rid of the issue.

milindur, 2018-08-27:
On which parameter change does the script crash? Which settings are visible, when entering the VTX menu? Is "Dev" set to None or anything else?

malelo1960, 2018-08-27:
Hello Christian,

first thanks for your reply. There are no settings visible. In the moment i switch to the vtx screen via menu the error message are there. Please see photo.

I have to switch the taranis power off to get rid of it. All other menu screens are ok.

Please excuse i didnt have a lot of experience with my x9d plus. I have it since 3 weeks.

Normal telemetry data, pids, filters and so on are all there. Only the vtx menu dont work. The S-Port from my frsky x4rsb is connected to the tx port on the vortex 150. 

UART2 is set to S-Port, Telemetry is enabled in Betaflight.

The betaflight version of the vortex 3.2.2. ( The last available from ImmersionRC)

 

With kind regards

 

Mario Löhl

milindur, 2018-09-01:
Are you using the latest version of the betaflight LUA scripts (v1.0.2)?

malelo1960, 2018-09-01:
Yes, im using the latest lua.script.

closed 2018-10-08 by stale[bot]