betaflight/betaflight-tx-lua-scripts #163: LUA script to set CLI parameters???

callagga, 2018-08-27:
Could I ask here please, is it possible to develop a LUA script for Taranis X9D+ / Betaflight that could be used to set CLI parameters on Betaflight?   

For example I had in mind to be able to set a switch to enable/disable "failsafe_procedure".  For example:  

 - Switch Position A: failsafe_procedure = GPS-RESCUE
 - Switch Position B: failsafe_procedure = DROP

_Just a quick check before jumping in to find out if it's possible (and there's not some reason I'm not aware of why it would not be possible)_

mikeller, 2018-08-27:
Not sure if putting the execution of CLI commands on a switch is a good approach:
- once CLI is entered, the flight controller can not be armed or flown any more before it is rebooted;
- as a consequence, the only way to make changes in CLI take effect is to save the new configuration and reboot;
- this means that only changing a switch while the craft is powered on, and connected to the TX will have an effect. Changing the switch while the craft is not connected, and then powering on / connecting the craft will result in the craft behaving as if the switch was in a different position to what it is.

callagga, 2018-08-27:
https://github.com/betaflight/betaflight/issues/6645

closed 2018-10-03 by stale[bot]